iotappstory/ESP-Library #19: Activation codes should be in code-font and properly spaced

msiebuhr, 2017-10-06:
It can be difficult to differentiate between `I`, `l` and a few other combinations due to the choice of font, ex:

[image]

This would be fixed by using a monospace/code font. Another option could be to avoid codes with ambiguous letters (one discussion with examples here: https://ux.stackexchange.com/questions/21076/are-there-numbers-and-letters-to-avoid-for-activation-codes-via-sms)

Also, when double-clicking the code (to mark the whole "word" for copy/pasting; I'm on a Mac), an extra space is included in front, which is rather hard to spot when entering into the devices' interface. Slight changes in markup can sometimes fix this; having a copy-to-clipboard button could be another way.

Onno-Dirkzwager, 2017-10-06:
Great feedback!
We updated it to Droid Sans Mono. As a windows user I did not experience the extra space you mentioned. I did however remove some space from the "source" which probably was the cause. Maby you could test it?

Onno-Dirkzwager, 2017-10-06:
Btw added your name to the thanks in the readme. Appreciate layout / markup. This gives us a head start for extending it!

msiebuhr, 2017-10-06:
Already saw the change, as I was setting up a device just how. Copy/pasting worked as a charm.

msiebuhr, 2017-10-06:
Also: Where should bugs in IOTAppStory itself go?

(Just found out a device can be in two projects at once, leading to some confusion about what app it should get...)

Onno-Dirkzwager, 2017-10-06:
Thanks this is already on our list! As the bug report page ;)

closed 2017-10-07 by Onno-Dirkzwager

msiebuhr, 2017-10-11:
@Onno-Dirkzwager can you point me to the bug-report page? I presume it's on iotappstory.com, but I cannot find any bug-reporting there.

Onno-Dirkzwager, 2017-10-11:
..on our list... so comming in the next update. If its lib specifik, please start a new issue. Otherwise we would appriciate an email to iotappstory@gmail.com