iotappstory/ESP-Library #22: Stored configuration gets erased with every new firmware

frog32, 2017-11-12:
Since the "eeprom" in an ESP8266 is in reality just some part of its flash storage, erasing it is always done in blocks (I think they are 4k). This means that the following only works if you have an entire block of configuration and in my case it erases the entire config.

https://github.com/iotappstory/ESP8266-Library/blob/master/src/IOTAppStory.cpp#L56

PotatoX, 2017-12-17:
I just added a pull request (#25) to allow for increading of EEPROM memory (up to 4K for ESP8266), and added some additional begin() flags to allow for more choice in how EEPROM memory is erased (or not erased) on the first boot.

I believe this will fix the issue.

SensorsIot, 2018-01-09:
In VirginSoil_Full example you find this line:
  IAS.begin(true,'P');   // 2nd parameter: Wat to do with EEPROM on First boot of the app? 'F' Fully erase | 'P' Partial erase(default) | 'L' Leave intact

In the initLoaders, we erase our part of the EEPROM completely to make sure we have no compatibility issues with earlier versions. Of course, you can change that. In all other sketches, we only delete it partially to make sure, that a completely different sketch can use the same space in EEPROM. The fields used by IOTAPPSTORY are not erased in this scenario.

closed 2018-01-23 by Onno-Dirkzwager