iotappstory/ESP-Library #35: DX not defined for many ESP8266 boards

tcottle, 2018-01-23:
dPIinConv expects that a series of data pin constants are defined (D0, D1, etc.)  These constants are not defined for many board types (Generic ESP2866 Module, Adafruit HUZZAH ESP8266)

```
C:\Users\XXXXXX\Documents\Arduino\libraries\ESP8266-Library-master\src\IOTAppStory.cpp: In member function 'int IOTAppStory::dPinConv(String)':

C:\Users\XXXXXX\Documents\Arduino\libraries\ESP8266-Library-master\src\IOTAppStory.cpp:728:56: error: 'D0' was not declared in this scope

   if      (orgVal == "D0"  || orgVal == "16")   return D0;
```

This issue was also present in 1.0.6

The issue may lie in the Arduino ESP8266 library.  Here is the pins_arduino.h file for the Adafruit board

```
/*
  pins_arduino.h - Pin definition functions for Arduino
  Part of ESP8266 core for Arduino - https://github.com/esp8266/Arduino
  Copyright (c) 2007 David A. Mellis
  Modified for ESP8266 platform by Ivan Grokhotkov, 2014-2015.
  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.
  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.
  You should have received a copy of the GNU Lesser General
  Public License along with this library; if not, write to the
  Free Software Foundation, Inc., 59 Temple Place, Suite 330,
  Boston, MA  02111-1307  USA
  $Id: wiring.h 249 2007-02-03 16:52:51Z mellis $
*/

#ifndef Pins_Arduino_h
#define Pins_Arduino_h

#include "../generic/common.h"

#define PIN_WIRE_SDA (4)
#define PIN_WIRE_SCL (5)

static const uint8_t SDA = PIN_WIRE_SDA;
static const uint8_t SCL = PIN_WIRE_SCL;

static const uint8_t LED_BUILTIN = 0;
static const uint8_t BUILTIN_LED = 0;

#endif /* Pins_Arduino_h */
```

Onno-Dirkzwager, 2018-01-24:
Hi tcottle,

Im affraid this is not an issue, rather a lack of a good explanation for why we have this function. We updated the [readme file](https://github.com/iotappstory/ESP8266-Library#dpinconv). Let us know if this answers your question.

tcottle, 2018-01-24:
Updated readme file looks great!  Pulled the latest ESP-Library and the change to dPinConv logic now allows complies w/o errors.  Tested using NodeMCU 0.9, Generic ESP8266 Module and Adafruit HUZZAH ESP8266 board definitions.

Onno-Dirkzwager, 2018-01-24:
Great to hear! And thanks for the feedback.

closed 2018-01-24 by Onno-Dirkzwager