iotappstory/ESP-Library #38: No retry?

vyruz1986, 2018-01-30:
It seems when the lib doesn't find the preconfigured WiFi at startup, it goes to config mode and waits forever.
Shouldn't it retry every x minutes, or scan continuously and switch back to normal mode when the configure network appears again?

Onno-Dirkzwager, 2018-02-01:
Hi vyruz1986,
Andreas introduced [preSetAutoConfig(false);](https://github.com/iotappstory/ESP8266-Library/blob/master/examples/VirginSoil-Full/VirginSoil-Full.ino#L64) to prevent going to config.

I dont know if this will help your project but you could change the setConfigPortalTimeout [here](https://github.com/iotappstory/ESP8266-Library/blob/master/src/IOTAppStory.cpp#L792)

As for "Shouldn't it retry every x minutes". I think this should be an option for devs to choose. But it is not currently.

As a workaround for your project. You could use the public function connectNetwork(); and build in a check at the end of your setup. Something like this:

if(WiFi.status() != WL_CONNECTED){
       connectNetwork();
}

I hope this works for your situation......and we have added the "scan" to the list of things we should have ;)

Btw. have you had any chance to test 1.1.0 ?

Onno-Dirkzwager, 2018-02-01:
[Max num of retries](https://github.com/iotappstory/ESP8266-Library/blob/master/src/IOTAppStory.h#L54) might be helpfull to.

closed 2018-03-17 by Onno-Dirkzwager