iotappstory/ESP-Library #39: Error: Unknown device. Please activate in config first!

tcottle, 2018-02-02:
I am having difficulty getting my modules recognized with the new library

These modules worked fine under 1.0.6

I have installed the 1.1.0 library and re-ran the  1.1.0 IASLoader sketch.  Re-entered Wifi credentials fine and activated with a new activation code.  On reboot I get the error message in the subject line

If I enter configuration mode by holding the GPIO 0 low the configuration web page loads fine.  Accessing on a PC using Chrome and the using the local IP address.  All buttons work except for the IOTAppstory button (which goes gray and stays gray).   Serial debug message *WM: [HTTP] GET... failed, try again

tcottle, 2018-02-07:
> tcottle, thankyou for testing!
>
> Theory: Your device has for whatever reason not saved the activation code.
>
> I cant replicate your situation. This is what I did:
>
>    - I ran the org 1.0.6 loader
>    - Started a test project. Updated to the example clock sketch (1.0.6)
>    - re-ran the 1.1.0 IASLoader sketch
>    - Added wifi & re activated with code
>    - Started a test project with a fresly compiled (1.1.0) demo sketch
>    - rebooted & succesfully downloaded the new firmware
>
> Maby you can help me figure out what I did or have different.
> I used:
>
>    - a Wemos mini
>    - Arduino IDE 1.8.5
>    - ESP8266 core for Arduino 2.4.0-rc2
>
> If your on Skype it might be easier to figure it out together. And post
> are results back here. Let me know...

I am out on travel for a couple of days and will get back on it Friday or
this weekend

I used:

   - Adafruit Huzzah
   - Arduino 1.85 through Microsoft Visual Studio
   - Esp8266 core for Arduino 2.4.0

I have 3 Huzzahs, one that hasn't been registered @ iotpappstory yet.  I
will test with that one using 1.1.0 IASLoader and report back my findings
Skyping this weekend is also an option

Onno-Dirkzwager, 2018-02-10:
Hi tcottle,
have your been using the ESP8266 core for Arduino 2.4.0-rc2?
We have had some buggy experiences with 2.4.0

tcottle, 2018-02-10:
2.4.0 -  will retest using 2.4.0-rc2

tcottle, 2018-02-12:
No joy.  Recompiled using ESP8266 2.4.0-rc2 and reuploaded my test apps 
1.1.0 loader does not download the new image.  I also tried it using the buttonloop method in one of my test apps.  Both report Error: Unknown device. Please activate in config first!

My next step is to re-verify that 1.0.6 still works for me(!)

Onno-Dirkzwager, 2018-02-13:
Thankyou for testing and your feedback!
There is obviously something major going wrong even if I can't reproduce it myself. Let's leave it here for the moment while we do further research. 

And yes re-verify 1.0.6 If this does not work out let me know and Ill help you back on board.

tcottle, 2018-02-13:
I reinstalled 1.0.6.  Still running ESP8266 arduino 2.4.0-rc2
Wiped my devices, apps and projects from IOTStory
Recompiled 2 apps.  One blinks a red LED and the other blinks a green LED
Loaded IOSLoader, reentered the credentials and activated the module (note
the IOTStory button needs to be renamed to something clearer like "Activate
Module")
I had a problem here.  I wanted to name the module "Huzzah Test 1".  The
website would not save the device using that name and I didn't see an error
message.  Renamed "Huzzah_Test_1" and all OK
I am running an Adafruit Huzzah.  Huzzah is not listed on the control panel
as a board type so I selected NodeMCU as the board type.  My apps were
compiled using NodeMCU 0.9 (ESP-12 Module).  I am pretty sure that is what
I did the last time the OTA feature worked
I loaded the Blink Green app to IOTStory and created a project
I loaded the Blink Red app into my test Huzzah and pressed the GPIO 0
button for a couple of seconds to trigger a firmware update

New error message!

Calling Home
 Current App: OTA_Blink_Red V0.3.1

 Checking for App(Sketch) updates from:
https://iotappstory.com/ota/esp8266-v2.php
 Downloading update... Error: Received Content-Length header is 0
 Checking for SPIFFS updates from:
https://iotappstory.com/ota/esp8266-v2.php
 Not Selected

I see one other issue.  When you enter an app you also manually enter the
version number.  I assume that must match this define statement Ex:
"#define VERSION "V0.3.1"   I don't think the SW gives a meaningful error
message in the case of a mismatch

Todd

Onno-Dirkzwager, 2018-03-17:
Hi  tcottle, sorry for the late reply, I mist out on the notifications.

Are you still having these issues or did you figure this out?

tcottle, 2018-03-17:
Hi I put it aside for a bit and intend to retest this weekend

Todd

Onno-Dirkzwager, 2018-03-17:
Great! 
Ill keep an eye on this issue, let me know how it went.

closed 2018-08-09 by Onno-Dirkzwager