iotappstory/ESP-Library #40: Exception (28) during WiFiManager Close operation

SandroAkira, 2018-02-08:
Using the IASLoader for v1.1.0, right after configuring the device on the WiFiManager, after choosing "Close" to close the WiFiManager the Wemos D1 Mini + OLED Shield generates an Exception and crashes. The dump message is generated bellow. 
For some reason the App configuration (besides WiFi config) is not saved as noticed in the attempts to update Firmware.
One important thing is that during the IOTAppStory configuration in the WiFiManager portal, the App name in the web appears blank and so the dropdown options. I was not able to make it visible by any means. This may be the cause for the crash.

[image]

==========================[BEGIN]===================================
[...]

*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Handle root
*WM: Server Close

Exception (28):
epc1=0x402385de epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000098 depc=0x00000000

ctx: cont 
sp: 3fff0fc0 end: 3fff17b0 offset: 01a0

>>>stack>>>
3fff1160:  40101be1 3ffee288 00000001 4021d016  
3fff1170:  402329f4 3fff2984 3fff30b4 3fff2bd4  
3fff1180:  3fff283a 00000000 00000000 0000011b  
3fff1190:  0000001a 018524b1 401021f2 00000100  
3fff11a0:  3fff283a 40104456 3fff2984 3fff2bd4  
3fff11b0:  00000036 3fff1c10 3fff2984 401043d1  
3fff11c0:  3fff2804 018524b1 00002200 4000050c  
3fff11d0:  3fffc278 40101f8c 3fffc200 3fff1c45  
3fff11e0:  3fff1c10 3fff1c10 3fff2804 4010453d  
3fff11f0:  40218e57 3fff2804 3fff283a 40218e60  
3fff1200:  00000008 3fff1a94 3fff12f0 08000000  
3fff1210:  60000200 08000000 08000000 3fff1a98  
3fff1220:  3fff1c45 00000003 3fff1c10 4021ec64  
3fff1230:  4025d400 00000020 003ff000 00000030  
3fff1240:  00000002 3fff2804 3fff12e8 4020cef2  
3fff1250:  00000000 00000000 00000000 3fff2848  
3fff1260:  3fff2804 3fff38e8 3fff1c10 4021eebc  
3fff1270:  3fff0000 3fff12e8 3fff1650 4020cf10  
3fff1280:  00000014 3fff38e8 3fff2804 4021f96c  
3fff1290:  ffffffff 00000000 3ffea971 00000008  
3fff12a0:  3fff38e4 000000ff 00000000 00000006  
3fff12b0:  3fff1b6c 00000000 00000000 401004d8  
3fff12c0:  00000000 4000422e 60000200 3fff1c10  
3fff12d0:  3fff38e4 3fff2874 3fff3894 4021f9ac  
3fff12e0:  3fff1c10 00000000 00000000 40218ef8  
3fff12f0:  3fff068c 000001c1 000001c1 4021f9d2  
3fff1300:  3fff1c10 00000000 00000000 402200d3  
3fff1310:  40106739 3fff2ae4 4025d400 4021d016  
3fff1320:  3fff1c10 00000001 3fff38e4 4021cc0c  
3fff1330:  00000b84 3fff38e8 3fff285c 3fff285c  
3fff1340:  00000000 00000000 3fff285c 00000000  
3fff1350:  00000000 00000004 3fff38e4 40218fd5  
3fff1360:  00000000 00000000 3fff38e4 4021975f  
3fff1370:  3fff068c 00000000 3fff3f1c 40219787  
3fff1380:  00000000 00000000 3fff2c9c 40203aef  
3fff1390:  4024d42e 00000000 3fff2c5c 40203ca0  
3fff13a0:  4024e5c3 00000001 3fff2c5c 4020d1fd  
3fff13b0:  3fff1db4 3fff0408 3fff2c5c 40206cdc  
3fff13c0:  00000001 3fff0408 3fff1650 402097de  
3fff13d0:  00000004 3fff26d8 00000004 00000000  
3fff13e0:  feefeffe feefeffe feefeffe 00000000  
3fff13f0:  3fff1410 00000000 3fff1650 3fff1410  
3fff1400:  3fff1410 00000000 3fff0098 402069e8  
3fff1410:  00000000 00000000 00000000 00000000  
3fff1420:  00000001 00000000 00000000 00000000  
3fff1430:  00000000 00000000 00000001 00000000  
3fff1440:  00000000 00000000 00000000 00000000  
3fff1450:  00000001 00000000 00000000 00000000  
3fff1460:  00000000 00000000 00000001 00000000  
3fff1470:  00000000 00000000 00000000 00000000  
3fff1480:  00000001 00000000 00000000 00000000  
3fff1490:  00000000 00000000 00000001 00000000  
3fff14a0:  00000000 00000000 00000000 00000000  
3fff14b0:  00000001 00000000 00000000 00000000  
3fff14c0:  00000000 00000000 00000001 00000000  
3fff14d0:  00000000 00000000 00000000 00000000  
3fff14e0:  00000001 00000000 00000000 00000000  
3fff14f0:  00000000 00000000 00000001 00000000  
3fff1500:  00000000 00000000 00000000 00000000  
3fff1510:  00000001 00000000 00000000 00000000  
3fff1520:  00000000 00000000 00000001 00000000  
3fff1530:  00000000 00000000 00000000 00000000  
3fff1540:  00000001 00000000 00000000 00000000  
3fff1550:  00000000 00000000 00000001 00000000  
3fff1560:  00000000 00000000 00000000 00000000  
3fff1570:  00000001 00000000 00000000 00000000  
3fff1580:  00000000 00000000 00000001 00000000  
3fff1590:  00000000 00000000 00000000 00000000  
3fff15a0:  00000001 00000000 00000000 00000000  
3fff15b0:  00000000 00000000 00000001 00000000  
3fff15c0:  00000000 00000000 00000000 00000000  
3fff15d0:  00000001 00000000 00000000 00000000  
3fff15e0:  00000000 00000000 00000001 00000000  
3fff15f0:  00000000 00000000 00000000 00000000  
3fff1600:  00000001 00000000 00000000 00000000  
3fff1610:  00000000 00000000 00000001 00000000  
3fff1620:  00000000 00000000 00000000 00000000  
3fff1630:  00000001 00000000 00000000 00000000  
3fff1640:  00000000 00000000 00000001 00000000  
3fff1650:  3fff00a4 3fff2c14 00000000 00000000  
3fff1660:  000668a0 00000000 00000a99 3ffe91bd  
3fff1670:  00000004 3fff26c4 3fff1674 3ffe9780  
3fff1680:  00000000 3ffe9780 00000000 3ffe9780  
3fff1690:  00000000 3ffe9780 00000000 3ffe9780  
3fff16a0:  00000000 3ffe9780 00000000 00000000  
3fff16b0:  ffffffff 00000001 3ffe9288 00000000  
3fff16c0:  01000035 00000000 00000000 00000067  
3fff16d0:  00000003 3fff0408 3fff06c8 4020ef4c  
3fff16e0:  a064a8c0 00ffffff 0164a8c0 4020efe8  
3fff16f0:  00000003 3fff0408 3fff0790 3fff0298  
3fff1700:  3fff1650 3fff1650 40210600 3fff0790  
3fff1710:  00000046 3fff06c8 3fff0098 00000001  
3fff1720:  00000046 3fff0298 3fff0098 40206a96  
3fff1730:  00000046 3fff0298 3fff0098 40206be5  
3fff1740:  3fffdad0 00000000 3fff0098 40202417  
3fff1750:  3fff1770 3fff0088 3fff0098 3fff0788  
3fff1760:  3fffdad0 00000000 3fff0098 402024fd  
3fff1770:  3fff233c 00000000 402023ec 4020238c  
3fff1780:  00000000 00000000 00000000 feefeffe  
3fff1790:  feefeffe 00000000 3fff0780 40210644  
3fff17a0:  feefeffe feefeffe 3fff0790 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

 EEPROM Configuration found

*-------------------------------------------------------------------------*
 Start INITLoader V1.1.0
*-------------------------------------------------------------------------*
 Mode select button: GPIO0
Boardname: INITLoader_********
 Automatic update: 0
*-------------------------------------------------------------------------*
 rtcMem
 markerFlag: U
 bootTimes since powerup: 9
 boardMode: N
*-------------------------------------------------------------------------*
 Connecting to WiFi AP
 . 
 WiFi connected
[...]

 Calling Home
 Current App: INITLoader V1.1.0

 Checking for App(Sketch) updates from: https://iotappstory.com/ota/esp8266-v2.php
 Error: Unknown device. Please activate in config first!

 Checking for App(Sketch) updates from: https://iotappstory.com/ota/esp8266-v2.php
 Error: Unknown device. Please activate in config first!

 Checking for SPIFFS updates from: https://iotappstory.com/ota/esp8266-v2.php
 Error: Unknown device. Please activate in config first!

 Checking for SPIFFS updates from: https://iotappstory.com/ota/esp8266-v2.php
 Error: Unknown device. Please activate in config first!

 Returning from IOTAppStory.com
*-------------------------------------------------------------------------*

                               ----------------   N O   A P P   L O A D E D   ----------------

----------------  P L E A S E  C R E A T E   P R O J E C T   O N   I O T A P P S T O R Y . C O M   ----------------

[...]

===========================[END]=====================================

SandroAkira, 2018-02-08:
After a couple of more tests I've noticed that besides the attempts to configure and save the Project in the WiFiManager interface, an error is issued in the debug messages:

```
[...]
*WM: WiFi save
*WM: Connecting wifi with new parameters...
*WM: previous settings invalidated
*WM: After waiting...
*WM: 3.80
*WM: seconds
*WM: Connection result: 
*WM: WL_CONNECTED
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: [HTTP] GET... failed, try again
[...]
```

Onno-Dirkzwager, 2018-02-09:
Thankyou for your detailed feedback! 
It seems your not the only one... Ill get back to you once we know more.

Onno-Dirkzwager, 2018-02-10:
Hi SandroAkira,
have your been using the ESP8266 core for Arduino 2.4.0-rc2?
We have had some buggy experiences with 2.4.0

SandroAkira, 2018-02-11:
Hi. In fact I'm using 2.4.0. How do I get the rc2?

SandroAkira, 2018-02-11:
Hi, just got the rc2. Now its working but there is another issue now. Probably something I am missing.
The script contacts Home but something is wrong with the header as noted bellow.
```
[...]
*WM: Handle root
*WM: Handle root
*WM: Server Close
 Connected... :-)
 EEPROM Configuration found

Configuration finished

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v0c897c37
~ld

 EEPROM Configuration found

*-------------------------------------------------------------------------*
 Start INITLoader V1.1.0
*-------------------------------------------------------------------------*
 Mode select button: GPIO0
 Boardname: INITLoader_**********
 Automatic update: 0
*-------------------------------------------------------------------------*
 rtcMem
 markerFlag: U
 bootTimes since powerup: 5
 boardMode: N
*-------------------------------------------------------------------------*
 Connecting to WiFi AP
 ...... 
 WiFi connected
 Device MAC: ***********
 Device IP Address: ***********
 MDNS responder started: http://INITLoader_***********.local

 To use MDNS Install host software:
 - For Linux, install Avahi (http://avahi.org/)
 - For Windows, install Bonjour (https://commaster.net/content/how-resolve-multicast-dns-windows)
 - For Mac OSX and iOS support is built in through Bonjour already
*-------------------------------------------------------------------------*

 Calling Home
 Current App: INITLoader V1.1.0

 Checking for App(Sketch) updates from: https://iotappstory.com/ota/esp8266-v2.php
 Downloading update... Error: Failed to verify the bin header
 Error: Header did not start with magic byte 0xE9

 Checking for SPIFFS updates from: https://iotappstory.com/ota/esp8266-v2.php
 No updates necessary!

 Returning from IOTAppStory.com
*-------------------------------------------------------------------------*

                               ----------------   N O   A P P   L O A D E D   ----------------

----------------  P L E A S E  C R E A T E   P R O J E C T   O N   I O T A P P S T O R Y . C O M   ----------------

[...]
```

Maybe there is something wrong when generating a new firmware.
I simply Export Compiled Binary through the Arduino interface and upload that file to the server.
Is this procedure correct?

Thanks!

Onno-Dirkzwager, 2018-09-12:
Hi SandroAkira,
im sorry you never got your answer! Somehow it got overlooked between al the other issues we handled. 

If your still interested we just released the 2.0.0 version.

closed 2018-09-12 by Onno-Dirkzwager