iotappstory/ESP-Library #45: No Issue, just a question

versamodule, 2018-02-23:
I am very much looking for ward to this. Will there be a port to the ESP32 ? 
I tried this with PlatformIO from this link here but to many errors (probably due to I was trying it with an ESP32). 
https://platformio.org/lib/show/1704/IOTAppStory-ESP8266/installation

Onno-Dirkzwager, 2018-03-17:
Hi versamodule,
sorry for the late reply. Currently the 1.0.6 release only supports the esp8266

However the upcomming 2.0.0 version does support the esp32. You are more than welcome to use this version. Just keep in mind that this still in beta and not officially released.

We did not start or maintain the platformio page you referred to. Make sure to get the latest version from the master here: https://github.com/iotappstory/ESP-Library

As most of our current users are on the Arduino IDE platform this is where we have done most of the testing. But apart from some minor board define issues it does work on platform io.

Ill try and send one of the users over here.

versamodule, 2018-04-02:
You can close it

closed 2018-04-02 by versamodule