iotappstory/ESP-Library #48: Receiving Update Failed. Error (-103): Forbidden (403)

pawisoon, 2018-03-17:
Hi,
I set up my board and attempt to update OTA but instead of fetching the new software I'm getting this.

```
 Calling Home
 Current App: IASBlink V1.0.0

 Checking for App(Sketch) updates from: iotappstory.com/ota/esp8266-v1.php
 Update Failed. Error (-103): Forbidden (403)
 Checking for App(Sketch) updates from: iotappstory.com/ota/esp8266-v1.php
 Update Failed. Error (-103): Forbidden (403)

 Checking for SPIFFS updates from: iotappstory.com/ota/esp8266-v1.php
 Update Failed. Error (-103): Forbidden (403)
 Checking for SPIFFS updates from: iotappstory.com/ota/esp8266-v1.php
 Update Failed. Error (-103): Forbidden (403)

 Returning from IOTAppStory.com
*-------------------------------------------------------------------------*
* button changed *
```

Can anybody help me fix this?

Onno-Dirkzwager, 2018-03-17:
Hi pawisoon,
I have had a look at the logs and you are missing the "device_sketch_md5"
This is nearly always caused by using an older core (sketch md5 came with 2.3.0)

I would suggest using 2.4.0-rc2 or later: https://github.com/esp8266/Arduino/releases

TIP! Make sure you don't have 2 versions (folders) in: C:\Users\*****\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266

This sometimes happens when installing with the Arduino boards manager. Especially when you installed the previous version manually.

Goog luck! And keep us updated

Onno-Dirkzwager, 2018-03-21:
Update use: 2.4.1
In my last reaction I did not realize it was already released.

Onno-Dirkzwager, 2018-03-28:
Any luck Pawisoon?
Can I close this issue?

pawisoon, 2018-03-28:
I didnt have time to test it. I will do that and let you know!

Onno-Dirkzwager, 2018-03-28:
Thanks for the update! No hurry, 

If you haven't tried yet, you might want to skip lib 1.0.6 
And go for the upcomming 2.0.0 [download from the master](https://github.com/iotappstory/ESP-Library)
(With the previously mentioned 2.4.1 core)

This has a lot more features and improved OTA updates & Device activation.

closed 2018-08-10 by Onno-Dirkzwager