iotappstory/ESP-Library #49: IAS update Error: "500 ESP MAC not configured for updates"

AlexZibin, 2018-03-20:
Hi, 
Unfortunately, I can't update the project in my NodeMCU board.

IASLoader compiles and loads fine, and everything goes as expected, until I want to make the OTA update of a sketch.

IAS logs say "500 ESP MAC not configured for updates" (please check at my IAS account "AlexKite").

I tried different versions of SDK with equal results.
Also tried different NodeMcu boards.

Please help to find the solution!

-------------------
Current ESP SDK version: 2.4.1 (I deleted previous versions before installing new ones)
Current Arduino IDE version: 1.6.12.
My IAS account: "AlexKite" 
-------------------

Onno-Dirkzwager, 2018-03-21:
Hi AlexZibin,
I checkt the logs and your settings. Everything seems ok. The usual suspects don't apply.
It looks like you may have had some type of error during registering your device. It is correctly added to IAS but maby the activation code failed to save to the esp.

I tried your scenario with a fresh board and could not recreate your problems.
I'm using: Arduino IDE 1.8.5 & 2.4.1 of the Arduino ESP8266 core.

Instead of trying to figure this out for 1.0.6, I would like to ask you to try the upcoming 2.0.0 version of our library. This has a different and more fail proof approach to device registration. Is greatly inproved and has many new features.

You will have to install it manually to your libraries folder (make sure to remove the 1.0.6 version)
Download it directly from the master: https://github.com/iotappstory/ESP-Library

And compile and upload your own IASblink app from the Arduino IDE examples folder. (the example apps on IAS are comiled for 1.0.6)

The only current documentation is the readme file in the master. We are currently working on the new wiki, a video and the website.

Good luck and give us an update about how it went

AlexZibin, 2018-03-21:
Hi Onno,
Thank you for your reply.
I will try ver. 2.0 tomorrow after work and will keep you informed.

AlexZibin, 2018-03-22:
Good evening Onno
Everything works fine with ver.2.0!
I will continue exploring on Saturday.

Thank you very much for your help!

Onno-Dirkzwager, 2018-03-22:
Good news!
Thanks for the feedback

closed 2018-03-22 by Onno-Dirkzwager