iotappstory/ESP-Library #50: Questions about Version 2.0

AlexZibin, 2018-03-27:
Hi Onno,
After brief testing of new version, I have some questions:
1) Now in initial Wifi connection webpage we cannot select a network from list. Only typing from scratch.  Old version was able to show a dropdown list of available networks. This was very convenient. 
Are you planning to make the same feature in ver. 2.0?
2) After connection and redirection, new webpage asks us to enter the login+pwd for IOTAppstory account. Previous version (with access code generation) was safer. Now login and password must be told to each connected device - is this safe enough?

Onno-Dirkzwager, 2018-03-28:
Hi AlexZibin,
good questions!

1.  initial Wifi connection
Yes this feature will return.

2. code generation & safety
We need you to login your "browser" to IAS so it can perform a few tasks for us over httpS in the background:

During initial activation: 
- Check if this device is already registered with IAS
- Get a list of all the boards we support for the pull down. 
- Try to figure out which board you have based on your compiled DEFINES, Flash size, core(8266 or 32) etc. And preselect  an option in the list if possible.
- Finish activation by sending device name and board type to IAS.
- If succesfull receive the **generated code** send it to your esp (check, confirm, save) and update the current page

_**Basicly do what you would previously do yourself in a browser**_

After activation / normal use
- Get, show and update projects & device settings on save

Previously most of the above was saved on and done by your esp. 
With lib 1.1.0 beta it became clear that 
- this took up to much resources
- and in most use cases your esp would not have enought free heap to do the above over httpS (your browser always has ;)

Once your device is registrered you only need to be logged in to use certain pages like the projects settings in Configuration Mode. (you can test this by going to our page, logout, browse to config etc.)

So a lot has changed this was only a small part of it. We are working on the documention and hope this will help when we release it.

I hope this a least answers your questions.

AlexZibin, 2018-03-28:
### Hi Onno,
Thank you for complete explanation of background processes!
I also tested [virginsoil demo](https://iotappstory.com/firmware-item/166/virginsoil-demo), it's really impressive.

**BTW**, I managed to **break** a __*Number Added Field*__ - I entered more than 8 digits in it. Please check its validation routine. Here's the value reported by ESP after saving the config:
##### Processing added fields
##### ID | LABEL     | LEN |  EEPROM LOC  | DEFAULT VALUE     | CURRENT VALUE                  | STATUS
##### 03 | Number  | 008 | 0528 to 0538  | 6                              | 1234567893я?Ѕ                 | OVERWRITTEN

### And again, thank you very much for your impressive work and prompt reply!

Onno-Dirkzwager, 2018-03-28:
AlexZibin,
glad you like it! And thankyou for testing and bringing up the nr of digits bug.

closed 2018-03-28 by Onno-Dirkzwager

Onno-Dirkzwager, 2018-04-06:
AlexZibin,
the list of available networks is back in! Redownload the master to update.

AlexZibin, 2018-04-06:
Wow! That's great!
Thank you Onno!