iotappstory/ESP-Library #51: WLAN password field too short

reibuehl, 2018-03-31:
The config variable "password" is defined as char password[STRUCT_CHAR_ARRAY_SIZE]; in IOTAppStory.h and STRUCT_CHAR_ARRAY_SIZE is set to 50 there while WPA2 passwords can be up to 63 characters long. 
Please define STRUCT_PASSWORD_SIZE = 64 and then use that length for the password.

Onno-Dirkzwager, 2018-03-31:
Reibuehl,
thanks for the feedback. Will do, in the meantime you can change it in the .h file.

reibuehl, 2018-03-31:
IOTAppStory.cpp also uses the define in three locations. I changed all to use a new define STRUCT_PASSWORD_SIZE but still can't connect and the debug output shows that only 49 chars are read. It seems that there is another limit on the HTML input field that I didn't find yet.

reibuehl, 2018-03-31:
The maximum field length is also used once in WiFiManagerMod.h. After changing that to 63 in the HTML code, it seems to work.

Onno-Dirkzwager, 2018-03-31:
Reibuehl,
which version of the lib are you using? 2.0.0?

reibuehl, 2018-03-31:
No, the 1.0.6 from the Arduino Library Manager.

Onno-Dirkzwager, 2018-03-31:
You might be interested in the upcoming lib 2.0.0
Its not in Lib Manager yet. You can download it straight from the master.
We are currently working on the documentation, website, video etc. So the best documentation we have currently is the readme file in the master.

Onno-Dirkzwager, 2018-04-01:
Hi Reibuehl,
I implemented your suggestion into the master (2.0.0)

closed 2018-04-01 by Onno-Dirkzwager