iotappstory/ESP-Library #55: Error: Received Content-Length header is 0 Reboot necessary!

dprice06468, 2018-05-27:
I’m using version 1.0.6 of the iotappstory library and 2.4.1 of the esp library. The IASLoader and IASBlink programs build successfully for my wemos D1 mini. I’ve uploaded the IASBlink program to IOTAppStory and loaded IASLoader to my D1 mini. When the D1 calls home to get an updated app, I get the following message repeated over and over. I checked the IAS logs and it show the request for the new app, but it never downloads and runs. Does anyone have any advice on how to correct this? Thanks!
-dave

-------------------------------------------------------------------------
Start INITLoader v1.2.0
-------------------------------------------------------------------------
Mode select button: GPIO0
Boardname: initloader-494
Update on boot: 0
-------------------------------------------------------------------------
rtcMem
bootTimes since powerup: 2
boardMode: N
-------------------------------------------------------------------------
Connecting to WiFi AP
.....
WiFi connected

Device MAC: Xxxxxxxxxx
Device IP Address: 192.168.1.166
MDNS responder started: http://initloader-494.local

To use mDNS Install host software:

For Linux, install Avahi (http://avahi.org/)
For Windows, install Bonjour (https://commaster.net/content/how-resolve-multicast-dns-windows)
For Mac OSX and iOS support is built in through Bonjour already
-------------------------------------------------------------------------
Calling Home

Checking for App(Sketch) updates from: https://iotappstory.com/ota/esp8266-v2.0.1.php
Download & Process update... Error: Received Content-Length header is 0 Reboot necessary!

ets Jan 8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

Onno-Dirkzwager, 2018-05-27:
Hi Dave,
it looks like you are mixing up releases:

1. The IASLoader that comes with lib 1.0.6 is version "V1.0.0" but you have "Start INITLoader v1.2.0" in your serial monitor.

2. The rest of the serial monitor feedback confirms this (different layout) & lib 1.0.6 updates from "esp8266-v1.php"

Our 2.0.0 lib is still in beta (released in a couple of weeks) But if I was you I would skip 1.0.6 and go straight to 2.0.0 (download and install from the [main page](https://github.com/iotappstory/ESP-Library)

Make sure you dont have multiple versions in: C:\Users\**user**\Documents\Arduino\libraries

What is you username at IAS? Ill follow your device in your next attempts to figure out if anything else is going wrong...

dprice06468, 2018-05-27:
Thanks so much!  I’ll work with the new version and let you know!  I
appreciate the help and the follow. My username is dprice.
-dave

dprice06468, 2018-05-28:
I've cleared out the old version from the libraries folder and have
downloaded the 2.0.0 version from your link.  As before i compiled the
IASBlink sketch and uploaded it to my project on IOTAppStory.  I then
uploaded IASLoader to my D1 mini.  I get the following error over and
over.  Looks like the invoked php version has been updated to 2.0.1 and the
init loader is still 1.2.0.  Is this the correct combination?  If it still
looks like an older version of some library is still referenced, can you
recommend steps to ensure it is removed completely?  thanks so much for
your help.  I love your site and the architecture.
-Dave

*-------------------------------------------------------------------------*
 Start INITLoader v1.2.0
*-------------------------------------------------------------------------*
 Mode select button: GPIO0
 Boardname: initloader-494
 Update on boot: 0
*-------------------------------------------------------------------------*
 rtcMem
 bootTimes since powerup: 40
 boardMode: N
*-------------------------------------------------------------------------*
 Connecting to WiFi AP
 .....
 WiFi connected

 Device MAC: xxxxxxxxxxxxxx
 Device IP Address: 192.168.1.166
 MDNS responder started: http://initloader-494.local

 To use mDNS Install host software:
 - For Linux, install Avahi (http://avahi.org/)
 - For Windows, install Bonjour (
https://commaster.net/content/how-resolve-multicast-dns-windows)
 - For Mac OSX and iOS support is built in through Bonjour already
*-------------------------------------------------------------------------*

 Calling Home

 Checking for App(Sketch) updates from:
https://iotappstory.com/ota/esp8266-v2.0.1.php
 Download & Process update... Error: Received Content-Length header is 0
Reboot necessary!

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

Onno-Dirkzwager, 2018-05-29:
Hi Dave,
"Received Content-Length header is 0" means the app / firmware you received is 0 kb, empty.
We initially searched in the wrong direction.... Due to a bug introduced 2 weeks ago some people had issues uploading their apps. Everything looked fine during upload. But the saved app ended up as an empty file.

We beleave we have fixed this issue and hope you can confirm this.

dprice06468, 2018-06-01:
Issue resolved!  thanks!

closed 2018-06-01 by Onno-Dirkzwager