iotappstory/ESP-Library #56: Device Mac not recognized under Arduino 1.0.6

fhassine, 2018-06-01:
Hi ,
I read and saw Andreas' video about the IOTAppstore, the idea is simply brilliant.

I made hours of tests in vain in the last two days, and always get the following message in logs:
"500 ESP MAC not configured for updates". In other words, the IOPAppStory doesnt see my project, or doesnt recognize my device. Am I right?
At the moment of execution, does the ESP contact the IOTAppStory site (via WIFI, since it has a connection)? Is the connection synchronous (http type) or is there a queue in place (MQTT type)?

I tried to compile with the version 2.0.0 (under branch developer under https://github.com/iotappstory/ESP-Library/tree/develop), by manually replacing the Master library by the new library. It compiles and loads well, but same result.
I would be greatful if anyone can respond.
Many thanks, Felix

Onno-Dirkzwager, 2018-06-02:
Hi Felix,
when you compiled ver 2.0.0 did you also use the new examples that came with this version?
And what is your username @ IAS? Ill follow your device and examine the logs.

fhassine, 2018-06-17:
> can we close this issue?

Hi Onno,

For some reason your emails went to spam box....I am sorry I didnt respond
to you yet.
For now, if it is a problem, I suggest you close the issue. I am developing
the project on my side and I have put the OTA topic secondary.
I thank you warmly for your involvement... I was thinking nobody was
responding.
I will come back to the topic when I restart on the OTA mechanism for
updating remotely the things.

All the best,
Felix

closed 2018-06-17 by Onno-Dirkzwager