iotappstory/ESP-Library #60: Making changes to app settings when no access to physical device

rcojoe, 2018-08-08:
Is it possible to make changes to the app when I don't have physical access to the machine?  
I have the VirginSoil_full demo loaded onto my device from the control panel.  I can make changes when I put the device in config mode my holding the button down.  

My question is: Is it possible to make changes to the settings from the control panel?  For instance, the time between calls home.

On the device page in the control panel,  when I click on edit next to my device,  it takes me to a blank control panel page.  Nothing is shown there.  Using 2.0 library on  a NodeMCU 1.0 12E

Onno-Dirkzwager, 2018-08-08:
Hi rcojoe,
at this moment no. But it is something we are considering.

Ill check the edit page. Thanks for your feedback.

Onno-Dirkzwager, 2018-08-08:
Rcojoe
No problems when I try to edit.
Do you have the same problem in the beta of the new version?

https://iotappstory.com/beta/cp/

rcojoe, 2018-08-09:
Beta works fine.  On the regular site, I logged out and back in but still get a blank (page header and footer but no body) device page.  

  The regular site has no log entries since yesterday after I went to the 2.0 library. This is a low priority now that I realized what the device edit page contains. I was thinking that is where I could set those local variables.  :)

closed 2018-08-09 by rcojoe