iotappstory/ESP-Library #66: v2 Beta: App keeps starting in config mode, not showing parameters on config page

vyruz1986, 2018-08-13:
I tried writing a very small app today on the v2 library and control panel which I received an invite for.
After loading the IASLoader.ino from the v2 library I was able to connect to my Wifi, link the device to my IAS account and setup a project with the app I uploaded.

The device seems to have downloaded the app successfully, since the status page of the device now displays the app and project name.

However there are 2 problems:
* I don't get to see the parameters I configured in my app using the `addField()` functions
* When I exit the configuration mode, the firmware seems to go into WDT reset.  I checked and the library is calling `ESP.restart()` and if I remember correctly that should not result in a WDT reset

This is my apps source code:
```
#define COMPDATE __DATE__ __TIME__
#define MODEBUTTON D3                                       // Button pin on the esp for selecting modes. D3 for the Wemos!

#include <ESP8266WiFi.h>
WiFiClient nsClient;

#include <PubSubClient.h>
PubSubClient MQTTClient(nsClient);

#include <IOTAppStory.h>                                    // IotAppStory.com library
IOTAppStory IAS(COMPDATE, MODEBUTTON);                      // Initialize IotAppStory

//MQTT Stuff
char* cMQTTServer = "";
char* cMQTTTopicPrefix = "";
char* cMQTTServerPort = "1883";
String strMQTTTopicPrefix;
unsigned long lLastMQTTConnectAttempt;
#define MQTT_CONNECT_INTERVAL 1000

//IAS Params
char* cCallHomeInterval = "86400";

//Bell params
bool bBellPressed = false;
bool bBellPressSent = false;

void setup() {
   Serial.begin(115200);

   //IAS Stuff
   IAS.preSetDeviceName("Doorbell");
   IAS.addField(cMQTTServer, "MQTTServeraddress", 16);
   IAS.addField(cMQTTServerPort, "MQTT Server port", 6);
   IAS.addField(cMQTTTopicPrefix, "MQTT topic prefix", 100);
   IAS.addField(cCallHomeInterval, "Call home interval (seconds)", 6);
   IAS.begin('P');
   IAS.setCallHome(true);
   IAS.setCallHomeInterval(atoi(cCallHomeInterval));
   IAS.callHome();

   MQTTClient = PubSubClient(cMQTTServer, atoi(cMQTTServerPort), nsClient);
   strMQTTTopicPrefix = String(cMQTTTopicPrefix);

   //Make sure MQTT Topic prefix starts and ends with a slash
   if (!strMQTTTopicPrefix.startsWith("/"))
   {
      strMQTTTopicPrefix = "/" + strMQTTTopicPrefix;
   }
   if (!strMQTTTopicPrefix.endsWith("/"))
   {
      strMQTTTopicPrefix = strMQTTTopicPrefix + "/";
   }

   MQTTConnect();

   Serial.printf("Doorbell pressed!");
   //The doorbell button will also reset the ESP from deepsleep, so we set bellPressed to true hardcoded
   bBellPressed = true;
}

void loop() {
   IAS.buttonLoop();
   MQTTClient.loop();
   if (bBellPressed) {
      if (!bBellPressSent) {
         GotoNormalSleep();
      }
   }
}

bool MQTTConnect() {
   if (!MQTTClient.connected()
      && millis() - lLastMQTTConnectAttempt > MQTT_CONNECT_INTERVAL)
   {
      lLastMQTTConnectAttempt = millis();
      if (MQTTClient.connect(String("ESP-" + ESP.getChipId()).c_str()))
      {
         return true;
      }
      else
      {
         Serial.printf("failed, rc=%i, try again in 1 second\r\n", MQTTClient.state());
         return false;
      }
   }
}

void GotoNormalSleep() {
   unsigned long ulSleepTime = atoi(cCallHomeInterval) * 1000;
   Serial.printf("Going to sleep for %lu seconds...\r\n", ulSleepTime / 1000000);
   yield();
   ESP.deepSleep(ulSleepTime, WAKE_RF_DEFAULT);
}
```

This is the serial output I get from the ESP:
```
*-------------------------------------------------------------------------*
 Start INITLoader v1.2.0
*-------------------------------------------------------------------------*
 Mode select button: GPIO0
 Boardname: initloader-968
 Update on boot: 0
*-------------------------------------------------------------------------*
 rtcMem
 bootTimes since powerup: 5
 boardMode: C
*-------------------------------------------------------------------------*
 Connecting to WiFi AP
 .....
 WiFi connected

 Device MAC: XX:XX:XX:XX:XX:XX
 Device IP Address: 192.168.x.x
 MDNS responder started: http://initloader-968.local

 To use mDNS Install host software:
 - For Linux, install Avahi (http://avahi.org/)
 - For Windows, install Bonjour (https://commaster.net/content/how-resolve-multicast-dns-windows)
 - For Mac OSX and iOS support is built in through Bonjour already
*-------------------------------------------------------------------------*

 C O N F I G U R A T I O N   M O D E
 STA mode. Open 192.168.x.x
 Exit config

Soft WDT reset

ctx: cont 
sp: 3fff2250 end: 3fff27e0 offset: 01b0

>>>stack>>>
3fff2400:  3fff16b8 000004e2 000004e2 4010020c  
3fff2410:  3fff5824 00000006 3fff5394 4010068c  
3fff2420:  000000ff 00000006 3fff5394 4020f810  
3fff2430:  a96d7c28 5d7a579a 1492b4f6 402060e3  
3fff2440:  3fff5824 00000006 3fff5394 4020723a  
3fff2450:  00000000 3fff574c 00000006 402214f0  
3fff2460:  00000100 3fff574c 3fff5829 40222110  
3fff2470:  00000001 00000014 9a44c18d cb9a9834  
3fff2480:  535cdd7f 140cccdf 98f32f84 3fff5829  
3fff2490:  00000004 00000000 3fff574c 402215b0  
3fff24a0:  00000004 00000000 3fff574c 40222587  
3fff24b0:  3fff5704 72a7e918 e5ca54fc 3fff5394  
3fff24c0:  00000000 600011f0 3fff561c 3fff5394  
3fff24d0:  00000000 00000004 00000004 40206ad8  
3fff24e0:  00000021 00001221 00000064 4020fa50  
3fff24f0:  00000050 00000000 00000031 3fff5829  
3fff2500:  3fff5829 00000004 3fff574c 3fff5829  
3fff2510:  00000004 00000004 3fff574c 40221ef0  
3fff2520:  00000061 3fff5829 3fff574c 402222cc  
3fff2530:  0000000f 3fff585c 3fff574c 401004d8  
3fff2540:  3fff2590 0000000e 00000010 00000000  
3fff2550:  00000000 3fff53cc 3fff574c 01000000  
3fff2560:  3fff2590 3fff531c 3fff574c 40222058  
3fff2570:  000001bb 3fff531c 3fff5374 3fff3cf4  
3fff2580:  0000a9ca 3fff531c 3fff5374 40206f95  
3fff2590:  3fff5374 00000000 3fff5394 40206586  
3fff25a0:  3fff5394 00000000 3fff26a4 00000000  
3fff25b0:  000001bb 3fff5374 3fff3cf4 00000000  
3fff25c0:  000001bb 3fff5374 3fff3cf4 4020731d  
3fff25d0:  3ffea3e8 c7c12e25 3ffea3e8 c7c12e25  
3fff25e0:  3fff16f8 00000000 3fff2670 40209e50  
3fff25f0:  00000000 3fff16d8 3fff262c 4010068c  
3fff2600:  3fff1111 00000000 3fff2670 4020a620  
3fff2610:  3fff10d8 3fff16d8 3fff2670 402031f0  
3fff2620:  00000000 00000000 00000000 00000000  
3fff2630:  3ffe9814 00000470 00000470 4010020c  
3fff2640:  3fff16f8 3ffe8e24 3fff26e0 3fff12d8  
3fff2650:  3fff16f8 3ffe8e24 3fff10d8 4020a6ce  
3fff2660:  3fff16f8 3ffe8e24 3fff10d8 40203276  
3fff2670:  3fff49a4 3fff5374 3fff3cf4 0000000f  
3fff2680:  0000000f 000001bb 3f011f40 3fff547c  
3fff2690:  0000001f 00000016 3fff5234 0000000f  
3fff26a0:  00000005 3fff54a4 0000016f 00000166  
3fff26b0:  3fff5334 0000001f 0000000f 3fff535c  
3fff26c0:  0000000f 00000000 00000000 00000000  
3fff26d0:  00000000 ffffffff 3fff1600 00000000  
3fff26e0:  00000000 00000000 00000000 3fff52e4  
3fff26f0:  0000002f 0000002d 3fff16f8 4020e65c  
3fff2700:  3fff16f8 3fff1448 3fff10d8 40204c6d  
3fff2710:  3fff3ba4 0f0aa8c0 40202930 4020fcd8  
3fff2720:  3fff1448 3fff16f8 3fff16f8 4020e5e8  
3fff2730:  3fff1448 3fff16f8 3fff1601 4020e60c  
3fff2740:  00000001 3fff11b3 3fff3d0c 40202d30  
3fff2750:  00000000 00000000 00000000 4024eda0  
3fff2760:  3fff12d8 3fff16f8 3fff10d8 40204d8c  
3fff2770:  00000000 00000000 00000000 4020f810  
3fff2780:  00000046 00000000 3fff10d8 402022df  
3fff2790:  3fff27b0 3fff10cc 3fff10d8 3fff17b8  
3fff27a0:  3fffdad0 00000000 3fff10d8 40202413  
3fff27b0:  3fff3a14 00000000 402022b4 40202218  
3fff27c0:  feefeffe 00000000 3fff17b1 4020fa1c  
3fff27d0:  feefeffe feefeffe 3fff17c0 4010070c  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vf6d232f1
~ld

*-------------------------------------------------------------------------*
 Start INITLoader v1.2.0
*-------------------------------------------------------------------------*
 Mode select button: GPIO0
 Boardname: initloader-968
 Update on boot: 0
*-------------------------------------------------------------------------*
 rtcMem
 bootTimes since powerup: 6
 boardMode: C
*-------------------------------------------------------------------------*
 Connecting to WiFi AP
 .....
 WiFi connected

 Device MAC: XX:XX:XX:XX:XX:XX
 Device IP Address: 192.168.x.x
 MDNS responder started: http://initloader-968.local

 To use mDNS Install host software:
 - For Linux, install Avahi (http://avahi.org/)
 - For Windows, install Bonjour (https://commaster.net/content/how-resolve-multicast-dns-windows)
 - For Mac OSX and iOS support is built in through Bonjour already
*-------------------------------------------------------------------------*

 C O N F I G U R A T I O N   M O D E
 STA mode. Open 192.168.x.x
```

Onno-Dirkzwager, 2018-08-14:
Hi vyruz1986,
Thanks for the feedback. I'm trying to replicate your situation. A few questions:
- which version of the arduido esp core are you using?
- which hardware? Wemos?
- are you able to compile and run the virginsoil_full example whithout the resets?

vyruz1986, 2018-08-15:
Sorry for the late reply, I have a couple of other things going on...

So I think it's not actually downloading the app, I made the wrong conclusion after seeing
```
Doorbell
Selected App
DoorbellSensor v1.0.0
```
in the config page. But when I start my device (Wemos D1 mini by the way) I still see the following:
```
 Start INITLoader v1.2.0
```

I reverted my commit for PR #67 just to be sure I didn't do anything wrong, but it's still going into WDT timeout when I load `IASLoader` and and try to let it download my Doorbell app.

So I took another Wemos D1 mini, and tried loading IASLoader, and setting up a test project with the IASBlink app. Unfortunately I get the same WDT timeout reboot.

So I started looking further, and found out I had the old boards manager URL for ESP8266 for Arduino still configured, I was stuck on 2.4.0-rc1, while in the meantime 2.4.2 had been released, so I updated to 2.4.2, now everything works fine...

Sorry for the false alarm, my bad :)

closed 2018-08-15 by vyruz1986