iotappstory/ESP-Library #71: After 5 days of not calling home,  can't get to check for updates

rcojoe, 2018-08-15:
Here is my log output:

*-------------------------------------------------------------------------*
 Calling Home

 Checking for App(Sketch) updates from: https://iotappstory.com/ota/esp8266-v2.0.1.php

 Checking for SPIFFS updates from: https://iotappstory.com/ota/esp8266-v2.0.1.php

 Returning from IOTAppStory.com
*-------------------------------------------------------------------------*

setup
```
  //IAS.begin();
  IAS.begin('P');                                     // Optional parameter: What to do with EEPROM on First boot of the app? 'F' Fully erase | 'P' Partial erase(default) | 'L' Leave intact

  IAS.setCallHome(true);                              // Set to true to enable calling home frequently (disabled by default)
  IAS.setCallHomeInterval(7200);                        // Call home interval in seconds, use 60s only for development. Please change it to at least 2 hours in production

  //-------- Your Setup starts from here ---------------
  yield();
  Blynk.config(auth);
  Blynk.connect(3333);  // timeout set to 10 seconds and then continue without Blynk
  while (Blynk.connect() == false) {
      Serial.println("Blynk Connecting");
    // Wait until connected
  }
  IAS.setCallHome(false);   
```
in my loop, I have a `IAS.callHome(); ` command that I trigger with a blynk button.  I know the code is running because the same button turns on an led, and in the log, it is calling home,  but there doesn't seem to be a response.

rcojoe, 2018-08-15:
'''
Start First Soil Joe v3.0.3
*-------------------------------------------------------------------------*
 Mode select button: GPIO0
 Boardname: virginsoiltest
 Update on boot: 1
*-------------------------------------------------------------------------*
 rtcMem
 bootTimes since powerup: 4
 boardMode: N
*-------------------------------------------------------------------------*
 Processing added fields
 ID | LABEL                          | LEN |  EEPROM LOC  | DEFAULT VALUE                  | CURRENT VALUE                  | STATUS

 01 | textLine                       | 016 | 0511 to 0529 | Light Show                     | Light Show                     | DEFAULT
 02 | Textarea                       | 080 | 0531 to 0613 | http://someapi.com/getdata.php?userid=1234&key=7890abc | http://someapi.com/getdata.php?userid=1234&key=7890abc | DEFAULT
 03 | Number                         | 008 | 0615 to 0625 | 6                              | 6                              | DEFAULT
 04 | Checkbox:Check me              | 001 | 0627 to 0630 | 1                              | 1                              | DEFAULT
 05 | Selectbox:Red,Green,Blue       | 001 | 0632 to 0635 | 0                              | 0                              | DEFAULT
 06 | Interval                       | 008 | 0637 to 0647 | 60                             | 60                             | DEFAULT
 07 | ledPin                         | 002 | 0649 to 0653 | 2                              | 2                              | DEFAULT
 08 | Timezone                       | 004 | 0655 to 0661 | 0.0                            | 0.0                            | DEFAULT
*-------------------------------------------------------------------------*
 Connecting to WiFi AP
 ....
 WiFi connected

 Device MAC: 5C:CF:7F:11:AB:E0
 Device IP Address: 192.168.1.14
 MDNS responder started: http://virginsoiltest.local

 To use mDNS Install host software:
 - For Linux, install Avahi (http://avahi.org/)
 - For Windows, install Bonjour (https://commaster.net/content/how-resolve-multicast-dns-windows)
 - For Mac OSX and iOS support is built in through Bonjour already
*-------------------------------------------------------------------------*
 Calling Home

 Checking for App(Sketch) updates from: https://iotappstory.com/ota/esp8266-v2.0.1.php

 Checking for SPIFFS updates from: https://iotappstory.com/ota/esp8266-v2.0.1.php

 Returning from IOTAppStory.com
*-------------------------------------------------------------------------*

'''
Above is the log after I hit the reset button on the board.  The log does not change after I power cycle the device.  It appears it is not getting out any longer but the blynk service is working so it is connected to the internet.

Onno-Dirkzwager, 2018-08-15:
rcojoe,
unfortunately this is caused by a changed certificate. Read more about it [here](https://iotappstory.com/news-item/18/to-all-beta-library-2.0.0-esp8266-users!)

And let us know if the mentioned approach works for you.

rcojoe, 2018-08-15:
Is it possible to change the fingerprint in the config page without recompiling?  Or was that option just added?

 

Joe

Onno-Dirkzwager, 2018-08-15:
This was just added. 
We expected an other 60 days and had planned to release it sometime next week. This was one of the last things on our very long to do list. And in retrospect we wish we had done it earlier.

rcojoe, 2018-08-16:
New code replaces  ias.buttonloop with ias.loop

Onno-Dirkzwager, 2018-08-16:
Yeah this was the last breaking change from 1.0.6 to 2.0.0. As this is more in line with other libraries and the underlying functionality it represents. (btn, timer, ota updates etc.) We should have informed the beta testers or left it in the develop branch untill the final release.

Are you back up and running and are your issues resolved?

rcojoe, 2018-08-16:
I am up and running with the new library.  I have not tried OTA update yet or set the fingerprint.  I’ll give it a test tonight.  I put the last comment in there just for a note to others if they stumble upon this ticket.

 

 

Joe

Onno-Dirkzwager, 2018-08-24:
Hi rcojoe, 
did you have time to do your final tests? Maby we can close this issue.

closed 2018-08-27 by Onno-Dirkzwager