iotappstory/ESP-Library #74: Missing constructor IOTAppStory with Appname and Version

Bolukan, 2018-09-16:
In examples the IOTAppStory object is initiated with the command:
`IOTAppStory IAS(APPNAME, VERSION, COMPDATE, MODEBUTTON);`

However, the current version of IOTAppStory got only the constructor for
`IOTAppStory IAS(COMPDATE, MODEBUTTON);`, see [Line 228](https://github.com/iotappstory/ESP-Library/blob/master/src/IOTAppStory.h#L228)
and functions for [preSetAppName](https://github.com/iotappstory/ESP-Library/blob/master/src/IOTAppStory.cpp#L94) and [preSetAppVersion](https://github.com/iotappstory/ESP-Library/blob/master/src/IOTAppStory.cpp#L101).

Onno-Dirkzwager, 2018-09-16:
Bolukan can you point out which examples, because I just had a look at all the new examples that come with the 2.0.0 library and cannot find the old: IOTAppStory IAS(APPNAME, VERSION, COMPDATE, MODEBUTTON);

Bolukan, 2018-09-16:
You are right. This only can be seen in PlatformIO. The 2.0.0 version is from 6 months ago, according to PlatformIO. Maybe PlatformIO updates only the files (in their cache) if the version has been updated? 
I will do some research how PlatformIO is updated.

Bolukan, 2018-09-16:
https://docs.platformio.org/en/latest/librarymanager/config.html

> PlatformIO Library Crawler uses library.json manifest to extract source code from developer’s location and keeps a cleaned library in own Library Registry.

Onno-Dirkzwager, 2018-09-16:
Ah we started prepairing this release +/- 6 months ago. 
So PIO pickt up that change directly from the json file in the master. Instead of Arduino lib manager that only looks at the releases. Good to know and thankyou for pointing that out.

Onno-Dirkzwager, 2018-09-16:
So to help the PIO users I think the easiest thing we can do is bump up the version number to lets say 2.0.01  
Do you agree and more important could you test this for us?

Bolukan, 2018-09-16:
That seems the most effective way. I will be glad to report back the behaviour in PlatformIO.

Issue #75 is also specific PlatformIO behaviour that blocks current use. Next to some warnings I succeeded uploading and using my first IotAppStory Project/App/Device combination.

Bolukan, 2018-09-18:
No effect. I guess library.json should be changed instead of library.properties
See https://docs.platformio.org/en/latest/librarymanager/config.html#version

Onno-Dirkzwager, 2018-09-18:
2.0.01 Done! Keep in mind that we will update to 2.0.1 in the near future. This update will include among other things your other issues.

Bolukan, 2018-09-18:
Manifest changed but examples did yet not change. I will check again after 24 hours ....

Bolukan, 2018-09-19:
No change. Platformio users can retrieve their examples from github. I will wait till 2.0.1 is released

Bolukan, 2018-09-19:
19-09-2019 19:00 CEST PlatformIO recognize v2.0.01.
Almost all examples are updated - at least I checked they have the right constructor - only WebAppToggleBtn is still old at this moment.

closed 2018-09-19 by Bolukan