iotappstory/ESP-Library #75: PlatformIO errors on pins the specific board is missing

Bolukan, 2018-09-16:
Compiling on PlatformIO for the **WeMos D1 R2 & mini** the compiler errors on 
```
#elif defined ESP8266_WEMOS_D1MINI || defined ESP8266_WEMOS_D1MINILITE || defined ESP8266_WEMOS_D1MINIPRO || defined ESP8266_NODEMCU || defined WIFINFO

		// DEBUG_PRINTLN("- Special ESP's -");
		if      (orgVal == "D0"  || orgVal == "16")   return D0;
		else if (orgVal == "D1"  || orgVal == "5")    return D1;
		else if (orgVal == "D2"  || orgVal == "4")    return D2;
		else if (orgVal == "D3"  || orgVal == "0")    return D3;
		else if (orgVal == "D4"  || orgVal == "2")    return D4;
		else if (orgVal == "D5"  || orgVal == "14")   return D5;
		else if (orgVal == "D6"  || orgVal == "12")   return D6;
		else if (orgVal == "D7"  || orgVal == "13")   return D7;
		else if (orgVal == "D8"  || orgVal == "15")   return D8;
		else if (orgVal == "D9"  || orgVal == "3")    return D9;
		else if (orgVal == "D10" || orgVal == "1")    return D10;
		else                                          return D0;
```
[Lines 1003-1017](https://github.com/iotappstory/ESP-Library/blob/4f2a4bc71c108c26ee5554d0c5f8226862110705/src/IOTAppStory.cpp#L1015)
as D9 and D10 are not defined for this board. (but called RX and TX).

Not knowing the function of this code, maybe this is not a good solution, but I propose to consider return 3 and 1 in stead?

Onno-Dirkzwager, 2018-09-16:
We never had compile errors while testing in Arduino IDE but you are [right](https://github.com/esp8266/Arduino/blob/master/variants/d1_mini/pins_arduino.h).

As you migh have noticed this function is also used in case of the [nodemcu](https://github.com/esp8266/Arduino/blob/master/variants/nodemcu/pins_arduino.h) and [wifinfo](https://github.com/esp8266/Arduino/blob/master/variants/wifinfo/pins_arduino.h)

So we will have to split this up.

As for why we have this basterd of a function check [this](https://iotappstory.com/wiki/arduino-esp/library-api-20/api-methods/dpinconv()) wiki page.

Bolukan, 2018-09-16:
Yeah, found the source code of the definition of the pins available for the compiler:

For variant "d1_mini", as the "WeMos D1 R2 & mini", "WeMos D1 mini Lite" and "WeMos D1 mini Pro":
https://github.com/esp8266/Arduino/blob/master/variants/d1_mini/pins_arduino.h

and variant "d1" as the "WEMOS D1 R1 (Retired)":
https://github.com/esp8266/Arduino/blob/master/variants/d1/pins_arduino.h

Onno-Dirkzwager, 2018-09-20:
Closing as you already Pull requested a solutions.

closed 2018-09-20 by Onno-Dirkzwager