iotappstory/ESP-Library #78: Enum for FieldType

Bolukan, 2018-09-24:
https://github.com/iotappstory/ESP-Library/blob/f8f234327a9163d2f2813f49ce9e937f2a17a93e/src/IOTAppStory.h#L80

I propose to replace the magic char with an enum for the fieldtype:
```
enum FieldType {
  textLine  = 'L',
  Textarea  = 'T',
  Number    = 'N',
  Checkbox  = 'C',
  Select    = 'S',
  Interval  = 'I',
  Pinnumber = 'P',
  timeZone  = 'Z'
};
```
Good idea?

PS: I am not expert enough to assess whether using a char-enum is the best solution (compared to ordinary enum + function to get char)

Onno-Dirkzwager, 2018-09-24:
At first glance this seems like a great idea. Ill have to think about the consequences....

As you probably know these chars are sent to your browser and a js scripts decides what to do with it.
Currently you are not locked in with a set ammount of options. 
And this is important for a future update where you as a developer can write js plugins for custom chars.

Bolukan, 2018-09-24:
Ok, I understand. hmmm

For the record: The code changes were succesful. I'll delete them. haha

closed 2018-09-24 by Bolukan

Onno-Dirkzwager, 2018-09-24:
It was a good idea! And as you keep having them, lets setup a Discord voice chat to talk details about where we are going.