iotappstory/ESP-Library #79: No Update on StartUp

GitLer90, 2018-10-03:
First of all, thank you for realising this great Project! =)

Since I have some devices "in the field" somewhere without WiFi, I want them only to update when I tell them to, for example via a Pin or Touchpad. 

I already set:
IAS.preSetAutoUpdate(false);
IAS.setCallHome(false); 
IAS.preSetAutoConfig(false);

But they still want to update when they are powered up. That takes away valueable seconds of battery time. 
So is there an easy way to tell them not to Update on StartUp? If not, could you please add this function to the library?

Thanks in advance!
Markus

CwlBroeders, 2018-10-03:
IAS.preSetAutoUpdate(false); Is intended to prevent any automatic update, also at powerup

Onno-Dirkzwager, 2018-10-08:
@GitLer90
As mentioned by @CwlBroeders the preSetAutoUpdate method is intented to en/dis able callhome for updates on boot.

I could not replicate your results. For me it works as intended. 
False: no updates on boot
True: update on boot

Did you place your preset before the IAS.begin() method?

Onno-Dirkzwager, 2018-10-15:
I'm closing this issue. Feel free to reopen it if you still have problems with the preSetAutoUpdate method.

closed 2018-10-15 by Onno-Dirkzwager

GitLer90, 2018-10-19:
Alright. I didn't know that IAS.begin() needs to be called after the presets. 
Now that works.

BUT

It still tries to connect to WiFi on Startup. And if that's not successfull it goes on with the program. But that takes like 10 s, so that problem's still there.
I could make it work by only calling IAS.begin() when I want to update in my code and doing nothing in the setup() part. But I guess that's not how it's supposed to work, right?