iotappstory/ESP-Library #83: Re-configure

alejandro-anv, 2018-10-18:
Is there any function call to force the device enter AP mode for re-configuring wifi even if it is already connected to one?

Onno-Dirkzwager, 2018-10-18:
Hi alejandro-anv,
their is no special function for this. So why not press the button for 7-10 sec?

alejandro-anv, 2018-10-18:
But... the config mode does not place the device in AP mode. 
I want to enter the mode it enters when it can't connect to wifi. 

Just tried espRestart() but...

```
syncrolights:129: error: 'espRestart' was not declared in this scope
     espRestart('C');
                   ^
exit status 1
'espRestart' was not declared in this scope
```

alejandro-anv, 2018-10-19:
Ok, but what I'm trying to do is to enter AP mode even if it is connected to an AP, because I want to change from the current AP to another.

Onno-Dirkzwager, 2018-10-20:
@alejandro-anv 
We do not support this but you bring up an interesting case.
Currently when using wifi multi (default) your device will connect to the strongest AP signal from the list.

We should add some sort of preffered AP selection in the next release. But this might take a while.
So Ill try to a figure out a temp workaround for your situation.
This will probaly use the onModeButtonVeryLongPress() callback and some custom code.

Ill let you know

alejandro-anv, 2018-10-22:
Do you know in which position of the nvram is located the network name/password?

Onno-Dirkzwager, 2018-10-22:
@alejandro-anv the config struct is stored in eeprom. I would suggest to add your second location as a second AP in config mode and use the very long press callback to force connect to AP2

alejandro-anv, 2018-10-27:
Hummm... what I want is some way to clear the current configuration... may be erasing the ap data from the eeprom. For doing this, I'm asking for the location of this data...

Onno-Dirkzwager, 2018-10-28:
@Onno-Dirkzwager 
This is a new issue! Anyway all is stored in eeprom in a struct:
https://github.com/iotappstory/ESP-Library/blob/master/src/IOTAppStory.h#L75

The size of the config struct depends on platform and defines. You can follow the serial monitor. Use the IAS.eepFreeFrom var. Or just set the begin method to ´F´
https://iotappstory.com/wiki/arduino-esp/library-api-20/api-methods/begin()

Onno-Dirkzwager, 2018-11-10:
@alejandro-anv
did you succeed in erasing your eeprom data?

Onno-Dirkzwager, 2018-11-26:
@alejandro-anv
Closing due to lack of response, please feel free to reopen if this is still an issue. Thanks.

closed 2018-11-26 by Onno-Dirkzwager