iotappstory/ESP-Library #84: Can't connect to configuration AP after another is configured

alejandro-anv, 2018-10-18:
I'v configured my device at home without problem. I connected to the ap with my phone, accessed 192.168.4.1 and all went ok. But Now I carried the device to other place.

 Checking the serial output I see it enteres configuration mode, but with the same phone I've configured it earlier I can't get the configuration page. Sometimes the AP is not listed. Sometimes the AP disappears when I'm connecting. Other times I can connect but it waits forever when trying to connect to 192.168.4.1. Other times it reports "ERR_ADDRESS_UNREACHABLE".

Any ideas? 

...........
 No Connection. Going into Configuration Mode
*-------------------------------------------------------------------------*
 Entering config mode.
*-------------------------------------------------------------------------*

 C O N F I G U R A T I O N   M O D E
 AP mode. Connect to Wifi AP "syncrolights"
 And open 192.168.4.1

Onno-Dirkzwager, 2018-10-18:
@alejandro-anv I have a device I use / test with at multiple locations. Office, home, colleague  etc. Without the mentioned problems.

It's clear this does not work for you. Can you provide some more details? Account name, device type and the phone / browser you use.

alejandro-anv, 2018-10-18:
Solved. The problem was I added a timer interrupt for blinking a led. It seems to mess with the AP software.

closed 2018-10-18 by alejandro-anv