iotappstory/ESP-Library #86: Feature request: Nextion Upload Support

GitLer90, 2018-10-22:
I am quite often using Nextion Displays in Projects with the ESP32.
So when I want to update the firmware of the ESP, that often includes new firmware for the Touchscreen.
It is already possible to upload the new compiled .tft-File to the Display via the ESP using the NexUpload.h from the official Nextion Arduino Library (When the file is already stored on the hardware/SD-Card in your system).
 
It would be an awesome feature for future releases, if you could not only upload a new .bin file to the ESP but also upload a new .tft file to the Nextion via your Webservice =)

Onno-Dirkzwager, 2018-10-23:
@GitLer90 
I think this is an awesome idea! We would need some help.
- which version do you have?
- do all versions support the nextupload.h?

GitLer90, 2018-10-24:
@Onno-Dirkzwager 
Sure I would like to contribute! =)

> which version do you have?

What do you mean? Arduino IDE? Nextion Editor? Itead Arduino Lib?

> do all versions support the nexipload.h?

Same thing^^

> on average how large do you expect the .tft files to be?

That depends mostly on how many pages and background pics you have. Mine are around 1-5 MB.

Onno-Dirkzwager, 2018-10-24:
@GitLer90 
- Which version Nextion Display do you have.
- Do all current Nextion displays support the nextupload.h method of uploading .tft files

GitLer90, 2018-10-27:
@Onno-Dirkzwager 
I currently use the 3.5" Basic Version (NX4832T035).
But yeah, all Nextion Displays support the NexUpload, since the Itead Nextion Arduino Lib is the same for every model. =)

Onno-Dirkzwager, 2018-10-27:
@GitLer90 
ok thankyou for the suggestion and information. I think this is a great idea. Ill leave this issue open and contact you somewhere in the future for testing, further questions, etc

Onno-Dirkzwager, 2018-11-11:
@GitLer90 
Good news! I received my Nextion and proved to myself in a test sketch that we can update it OTA using IAS. It will take an other few weeks to implement it correctly in our library and web service. But it is coming!

Onno-Dirkzwager, 2018-11-18:
@GitLer90 
what is your username at IAS? Ill add a nextion upload field to your account so you can test ota update to your screen.

GitLer90, 2018-11-20:
Wow, great news! 
My username: KIMETO

thank you!

Onno-Dirkzwager, 2018-11-20:
@GitLer90 do you have an 8266? or only 32's

GitLer90, 2018-11-21:
@Onno-Dirkzwager Only ESP32's

Onno-Dirkzwager, 2018-11-22:
@GitLer90 I hope to have the ESP32 version ready by friday

Onno-Dirkzwager, 2018-11-23:
@GitLer90 Which nextupload library are you using on the esp32?

alighadyali, 2018-11-25:
This would be awesome.

My username is alighadyali 

Thanks!

Onno-Dirkzwager, 2018-11-25:
I have run into some problems and out of time while developing the ESP32 version for OTA updates.

For those using the ESP8266:
https://github.com/iotappstory/ESP-Library/blob/nextion-beta

GitLer90, 2018-11-26:
@Onno-Dirkzwager I am using the Nextion Library provided my Andreas Spiess for the ESP32:
https://github.com/SensorsIot/Nextion-Enhanced/tree/master/ITEADLIB_Arduino_Nextion

GitLer90, 2018-11-26:
> I have run into some problems and out of time while developing the ESP32 version for OTA updates.
> 
> For those using the ESP8266:
> https://github.com/iotappstory/ESP-Library/tree/nextion-beta

@Onno-Dirkzwager Hm, does that mean the ESP32 Version will take longer?
I don't have ESP8266s to test it.

Onno-Dirkzwager, 2018-11-26:
@GitLer90 
So you don't use hardwareserial for the esp32.  Which softwareserial library do you use?

GitLer90, 2018-11-26:
@Onno-Dirkzwager The lib uses Serial2 for communication with the Nextion. And that's HardwareSerial as far as I know. It's on Pin 16 and 17 of the ESP32

Onno-Dirkzwager, 2018-11-26:
@GitLer90 
I asumed sofwareserial as this was used in the SensorIOT / Aruino_Nextion upload.h example.
Software serial is also used on all the ESP8266 examples I studied to make the OTA updates happen.

Their are a few software esp32 forks. But they did not work for me. And since the esp32 has 3 serial ports I opted to go back to hardware serial (as the ITead lib org started of as) 

I spent this weekend on getting the ota updates to use hardwareserial on the ESP32. But was unsuccesfull.

I did not give up! Just ran out of time.

Onno-Dirkzwager, 2018-11-26:
@GitLer90 @alighadyali 
I added the Nextion upload fields to the "My Apps" page. 
You can add .tft files below the sketch and spiffs files.

GitLer90, 2018-11-27:
@Onno-Dirkzwager Hm I guess you can also use SoftwareSerial on actual HardwareSerial Pins. Even that doesn't really make sense. But I remember I did that once with an Arduino Nano, because the libs expected SoftwareSerial and it worked.

I tried to upload .tft files to my ESP32 via IAS but it didn't change anything for the Nextion Display. I guess that's because of the problem you described above. Let me know if I can help you somehow.

Onno-Dirkzwager, 2018-11-28:
@GitLer90 I need your help here.....
As you know I got it working on the esp8266

But on the ESP32 I can't get anything to run on the Nextion. Not even a simple load bar....
It would be of great help if you sent me / pointed me to an example sketch and library that you use yourself and works so we can compair.

GitLer90, 2018-11-29:
@Onno-Dirkzwager 

I created a very simple Example .tft-file and Sketch for testing the communication between the ESP32 and the Nextion Display. Comments are found in the .ino file.

https://www.dropbox.com/sh/41enq2xpa0jad91/AADipiG6olQwdvRRy1i-MNqBa?dl=0

If that still doesn't work for you, it's maybe a problem in the libs. Also I am not entirely sure if I edited the HardwareSerial.h file. 
Feel free to text me your test results. 
PS: Maybe we should chat about that elsewhere to not blow up this issue ;)

Onno-Dirkzwager, 2018-11-29:
@GitLer90 
Thankyou, maby you can come onto our discord or add me on skype: mentalsolutions

abs-zero-ian, 2018-12-03:
Hi Onno, from testing the 8266 version (with the extra yield() just before initiating the connection to the display, I have seen a couple of issues:

1) Following multiple uploads, the SPIFFS is reported to be full, which causes an error in (I think) the ESP NexUpload lib (insufficient space) while transferring the file from IAS.

2) Sometimes the sketch crashes at the point of beginning upload to the display from the ESP.  The cause seems to be 'Soft WDT' so I'm looking for a good place to insert another yield() at this point.

The yield() seems to improve the chances of being able to update the Nextion display multiple times, but when the sketch reboots, the display hangs as you observed (and needs power cycling). I seem to get to about 7 or 8 updates without physically touching the display before the crash occurs. 

If it helps I will experiment with serial.print() debug messages to see how far it gets through the NexUpload lib and also display the free space at the beginning of each call.

I thought it might help if I share this here do that anyone else who may be looking at this might also see what is happening.

Really useful extension to IAS, hope this helps.

Onno-Dirkzwager, 2018-12-03:
I updated the nextion-beta branch
- removed SPIFFS storage all together
- direct serial upload to the Nextion from the wifi client
- added yields on top of most functions

abs-zero-ian, 2018-12-08:
Thanks Onno, I've been testing again, just using the VirginSoil example and just effectively the update process.

By and large it seems to work pretty well now but I've seen a couple of instances where the display seems to have corrupted data, I'm not certain, but I think I can reproduce the cause.

Now that SPIFFS isn't used in the update process, the update upload to the display starts pretty much immediately (very welcome!).  

I believe I caused corruption of the upload by uploading a new Nextion firmware onto the IAS control panel while the update was taking place.  I thought I should raise this in case the process is used also in sketch upload, which would be less easy to recover?

I'll start playing with this in a real sketch now. :+1:

abs-zero-ian, 2018-12-29:
OK, so I've been using this for a couple of weeks now and today seem to be getting some issues, which look more like IAS is getting confused than the sketch (I'm sure I'm wrong, but I'll try to explain what I do and how I end up fixing it).

I have one active project that uses Nextion and I've been developing it for a while, for the most part not touching the TFT file and just uploading new sketches.  Today, when replacing the sketch .bin file on IAS, on the next reboot, I get the Nextion 'white screen' briefly showing it is being uploaded to, before a message saying the screen type doesn't match and a (screen) hang.  The sketch then continued as usual, albeit with a non-working screen.

Any subsequent reboot (removal and reinstatement of power) of the target went back to the same state (i.e. trying to get the TFT file again) and all the time, the screen would power up with the existing TFT file still present.

The only way I could fix it was to delete the project and app and start again in IAS, I didn't delete the device.

Some observations your consideration:

1. It isn't possible to delete a Nextion firmware once it is on IAS (while it disappears from the App pane, it returns when you reopen the app pane)
2. IAS forgets that there is no Nextion firmware and defaults to a blank entry (requires the user to delete the entry every time a sketch is uploaded)
3. When it was getting 'confused' IAS reported that the sketch either didn't exist or was incompatible.

Hopefully this all helps in some way. :-)

Onno-Dirkzwager, 2018-12-31:
@abs-zero-ian This nextion beta uses a different update address as the normal version. And I think your errors have something to do with me editing this file and adding md5 support for file verification (is an update necessary or not)

Onno-Dirkzwager, 2019-01-06:
@GitLer90 @alighadyali @abs-zero-ian @msiebuhr
@Nredor accepted my pull request. You can now find the ESPNexUpload in the Arduino library manager!

JustusW, 2019-01-09:
Since there's no official repo for the page: I'd really appreciate if the Nextion upload thingy didn't require me to delete the superfluous upload field every time I change my app.

abs-zero-ian, 2019-01-09:
I agree - I've raised this on the Discord channel and it is probably on the
list somewhere. :-)

Onno-Dirkzwager, 2019-01-11:
Hi @JustusW,
we fixed the version trash button (that forgot to delete on save)

Im not sure if this solves your issue let me know!?

JustusW, 2019-02-03:
Hi @Onno-Dirkzwager,

It sort of did help, I still get the empty field and can't submit until I wrote something in there or deleted it, but it will at least stay filled in on subsequent submits. If you have any way of releasing your web code on Github I would happily assist you in evolving it forward.

Onno-Dirkzwager, 2019-02-11:
@JustusW 
saw this last message a little late. But this is also fixed as I ran into it as wel.

This nextion field was for the beta testers. And this part of the page will be replaced in the future.

GitLer90, 2019-02-14:
Thank you! It works now! =)

closed 2019-02-14 by GitLer90

joeltbennett, 2019-10-31:
@Onno-Dirkzwager Hi, I've been using the latest NexUpload (ESP32) library that you contributed to and I'm having a problem that I'm hoping you can help me with.

I don't use this appstory thing (though it seems cool), I have a server that works fine for GET requests. I update my ESP32 just fine, and I have the NexUpload library connecting and downloading an 8.5mb TFT file perfectly - except when it gets to the very end.

The download reaches 100% and then a sort of File Check thing begins, and that goes to 100% immediately - and after that, I get an error:

**Update Failed: check Error!**

It looks like it's trying to give me a readout of the error, but it's not visible because there's a text field in front of it that won't disappear (this is all in the UART screen mode).

Anyway, I'm really set on getting this thing to work because it'll open up a lot of doors for my project. Have you had any issues like this before? Is it because the file size is 8.5mb?

I'm using a 5" Enhanced screen ([NX8048K050](https://nextion.tech/datasheets/nx8048k050/)) and an ESP32 WROOM chip (WeMos R32). The baud rate has been set to 15200 on the Nextion, so I'm using that for all my serial dealings. Finally, I'm using the HTTP GET to serial stream method outlined in the NexUpload/esp32 lib docs.

Any help here would be greatly appreciated, and paid for if needed ^_^