iotappstory/ESP-Library #87: Problems with Download & Process update procedure

MrSuhov, 2018-10-25:
Hello!

Suddenly updates stop working properly.
Usually, after successfull downloading of a new firmware my esp32 reboots with it.
But yesterday I've got the following:

```
*-------------------------------------------------------------------------*
 Calling Home

 Checking for App(Sketch) updates from: https://iotappstory.com/ota/esp32-v1.php
 Download & Process update...
 Checking for SPIFFS updates from: https://iotappstory.com/ota/esp32-v1.php
 Not Selected

 Returning from IOTAppStory.com
*-------------------------------------------------------------------------*
```

and the sketch debug messeges show that the firmware was not really updated.

If I reboot ESP manually it does not see any firmware updates:
```
*-------------------------------------------------------------------------*
 Calling Home

 Checking for App(Sketch) updates from: https://iotappstory.com/ota/esp32-v1.php
 No updates necessary!

 Checking for SPIFFS updates from: https://iotappstory.com/ota/esp32-v1.php
 Not Selected

 Returning from IOTAppStory.com
*-------------------------------------------------------------------------*
```

What could be the cause for such a behavior?

MrSuhov, 2018-10-25:
Juat a little more info.

Behavior of initial IASLoader is the same:

```
 Calling Home

 Checking for App(Sketch) updates from: https://iotappstory.com/ota/esp32-v1.php
 Download & Process update...
 Checking for SPIFFS updates from: https://iotappstory.com/ota/esp32-v1.php
 Not Selected

 Returning from IOTAppStory.com
*-------------------------------------------------------------------------*
 If mode button is released, I will enter in firmware update mode.
*-------------------------------------------------------------------------*
 If mode button is released, I will enter in configuration mode.
*-------------------------------------------------------------------------*
 Calling Home

 Checking for App(Sketch) updates from: https://iotappstory.com/ota/esp32-v1.php
 No updates necessary!

 Checking for SPIFFS updates from: https://iotappstory.com/ota/esp32-v1.php
 Error: Max. callhome / min / device reached!

 Returning from IOTAppStory.com
*-------------------------------------------------------------------------*
```

Onno-Dirkzwager, 2018-10-25:
@MrSuhov this seems to be a problem with the update flag on the esp32 side of things. We use this flag as the esp32 core does not support the sketchmd5 method yet for compairing sketches.
Please reupload your sketch to reset the flag (you dont have to recompile) and let us know if this works.

I personaly had something similar last week. We will add a reset flag option for esp32 devices.

MrSuhov, 2018-10-31:
A few updates on this.
Since I can easily reproduce the issue I could use the low level debugging like in ESP8266 with SSL+HTTP_CLIENT to get more info. However I cannot find this option for esp32 in the latest arduino IDE.

When fails it takes 5-7 sec to comunicate with IOTAppStory:
```
14:29:23.963 ->  Checking for App(Sketch) updates from: https://iotappstory.com/ota/esp32-v1.php
 Download & Process update...
14:29:30.928 ->  Checking for SPIFFS updates from: https://iotappstory.com/ota/esp32-v1.php
```

However on success it takes no more than 30 sec:
```
14:32:47.289 ->  Checking for App(Sketch) updates from: https://iotappstory.com/ota/esp32-v1.php
 Download & Process update..... Received & installed: ClimateApp esp32 test 1.0.4 Reboot necessary!
14:33:21.151 -> ets Jun  8 2016 00:22:57
14:33:21.198 -> 
14:33:21.198 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

So it looks like it cannot complete firmware download. 
I have two WiFi access points: 1) 3G - tethered WiFi and 2) a regular WiFi router.
The statistics is different:
1) 100% fail using a 3G - tethered WiFi (iPhone personal hotspot)
2) 15% fail using a regular WiFi from router

MrSuhov, 2018-10-31:
And thanks a lot for this great new feature in 2.0.1 - onFirmwareUpdateSuccess();

Onno-Dirkzwager, 2018-11-05:
@MrSuhov still having problems? And did the mentioned (Please reupload your sketch to reset the flag) have any effect?

MrSuhov, 2018-11-05:
Yes, the problem still persists. And yes, reuploading sketch many times make it possible to collect the statistics above. I will get back later with the dependency of fail % upon the average ping response time from iotappstroy.com. My best guess is that poor internet connection causes all this.

And having the _onFirmwareUpdateSuccess()_ function now I can save to flash the number of FW version and if the update was initiated and version stays the same it will report the fail.

I do not use a regular FW check from the iotappstory, I am doing FW check just once - on esp startup (I have possibility to reboot esp remotely). So the problem is not that big - I can reboot esp as many times as needed to finally get the update. All i need is to implement the mentioned version change track.

Onno-Dirkzwager, 2018-11-05:
@MrSuhov 
as you have been the only one with this problem (that we know) and we both suspect the internet connection. Ill close this issue. But feel free to reopen!

closed 2018-11-05 by Onno-Dirkzwager