iotappstory/ESP-Library #88: Captive portal config and http(s) issues

yknivag, 2018-11-05:
Hello, apologies if someone has asked these questions or there are details somewhere in the documentation, I couldn't find anything.  I posted this in Discord but no-one replied there so I have posted here also.

Is it possible to set the IAS configuration website to be a "captive portal" when the device is in AP mode?  So that it behaves like WiFiManager and prompts for sign-in?

Also is there anyway to use HTTPS but to fall back to HTTP if that fails?  Or is HTTP/HTTPS only a compile time option with the #define?  Either that or to set HTTPS to retry without cert verification if it fails?  Only if there isn't a way to fail-over then if the SSL certificate ever changes it will no longer be possible to update automatically (especially on ESP32 for which there is an embedded RootCA in the code) rendering the device stuck until it can be retrieved for manual re-programming

I know that the RootCA will last longer than the site cert but that is only relevant if any replacement/renewed site cert is issued against the same root and that isn't always the case with many providers - many providers start issuing against new root CAs significantly before the expiry of the old to ensure a smooth crossover and some large providers issue against a pool of roots to minimise the impact of having a revoke one.

One final question - is it possible to use IAS for a project build with the full ESP-IDF rather than in the Arduino IDE?

The project I'm intending for this will use an ESP32 and be built outside the Arduino environment as that doesn't yet support the required ESP-ADF.  The project is intended for use by people with severe visual disability and so reliable remote updates with no user interaction are important.

Hopefully someone can help?

Onno-Dirkzwager, 2018-11-05:
Hi @yknivag
I must have missed you on Discord. Good questions!

- For the config pages. We made an hybrid cloud webapp. This gets the css, js and most of html pages from the IAS "cloud". And first time devices also use the browser to setup an httpS connection for device registration. (your browser acts as a relay between IAS and your esp) When in "captive portal" mode the browser only has acces to the esp and not to the internet. In a future release we will give you the option to use a static SPIFFS version of the config pages which might (depends on features) be able to run in the "captive portal"

- Currently there is no fallback. You are right about the compile time option with the #define. The esp8266 config pages have a page for updating the fingerprint. For the esp32 I'm currently working on a config page which lets you update the RootCA and save it to spiffs. (maby you could help by testing a beta in a couple of days?)

- ESP-IDF? Not yet! We started out with the Arduino IDE as most of our users are here. 2.0.1 made it compatible for compiling with PlatformIO. ESP-IDF requires a full port of the library. Our ota server however does not care wat kind of device "platform" calls home. As long as it provides the necessary headers and accepts .bin files in return. If you are already working in the ESP-IDF and can initiate an ota update we will help you get it compatible with our services! Just ask.....

Sounds like you have an interesting project! I have had several projects trying to help disabled people. And would like to offer my help where ever you need it.

I hope my answers made sense and help!?

yknivag, 2018-11-05:
Hi Onno,

Many thanks for getting back so quickly!

Regards the captive portal I was thinking of when the ESP cannot connect to any WiFi and it enters AP mode and one must connect to it and browse to 192.168.4.1 to set up the WiFi.  The browser then has no internet access as it is only connected to the ESP - it is in this condition that I'd like to be able to have a captive portal to make setting the WiFi credentials easier.

No fallback makes https a no-go for me then.  Would be good to have the option to turn off client.verify() to get around that.  Otherwise the risk of the project needing physical attention is too great.  Even if the root CA is in SPIFFS that still doesn't help really as access is still required to update it in SPIFFS and a call home for a SPIFFS update would fail due to the root CA it was trying to get so physical access would still be needed.

Well if it works in PlatformIO then it may be possible - apparently it is possible to use the IDF and, with modifications, the ADF in PlatformIO.  I don't know - this is the first time I have had to deviate from the Arduino IDE and I am still very much at the conceptual research stage.  IAS just seemed a fascinating way of dealing with upgrades.

Onno-Dirkzwager, 2018-11-05:
@yknivag 
**Captive portal**
We could consider a define for this. But it would only the "simple" wifi manager you got the first time you added a device. This would be a define specificly for your type of situation, where you want this behavior. As most users don't want to leave the portal and have to browse to the ap for the other options. (currently we can automaticly redirect you to the eps's ip)

**Fallback**
We had a fallback in previous versions but it consumed to much memory having both versions "ready". 
We will consider modifying the current define from (https / http) to (https / http / both).

**cert update**
This will actualy be done by your browser (as it has https...) automaticly when you enter config mode. A js script gets the cert over https. And sends it to the esp.

**ESP-IDF**
Full support with a special library will take an other +/- 6 months. But as mentioned if you are willing to modify one of their ota update examples it will work with IAS. (I don't know enough about using both IDF & ADF in PIO at the same time. But keep me posted)

**fascinating way of dealing with upgrades**
Thankyou! I hope my answers will make you consider to keep using IAS!

Onno-Dirkzwager, 2018-11-05:
@yknivag 
**Captive portal**
We will consider a define for this. But it would only the "simple" wifi manager you got the first time you added a device. (or de static spiffs version that comes in the next release)

These version of the config pages do not have the fancy fade / swipe / timezone / pin settings stuff the current cloud version has. But is seems you don't need this anyway.

This would be a define specificly for your type of situation, where you want this "Captive portal" behavior. As most users don't want to leave the portal and have to browse to the ip for the other options. (currently we automaticly redirect you to the eps's ip if your in the browser connected etc.)

**Fallback**
We had a fallback in previous versions but it consumed to much memory having both versions "ready". 
We will consider modifying the current define from (https / http) to (https / http / both).

**cert update**
This will actualy be done by your browser (as it has https...) automaticly when you enter config mode. A js script gets the cert over https. And sends it to the esp.

**ESP-IDF**
Full support with a special library will take an other +/- 6 months. But as mentioned if you are willing to modify one of their ota update examples it will work with IAS. (I don't know enough about using both IDF & ADF in PIO at the same time. But keep me posted)

**fascinating way of dealing with upgrades**
Thankyou! I hope my answers will make you consider to keep using IAS!

Onno-Dirkzwager, 2018-11-15:
@yknivag any feedback?

Onno-Dirkzwager, 2018-11-26:
@yknivag 
Closing due to lack of response, please feel free to reopen if this is still an issue. Thanks.

closed 2018-11-26 by Onno-Dirkzwager