iotappstory/ESP-Library #93: Error: Max. callhome / min / device reached! and magic Byte 0xE9 error

Cloolalang, 2018-12-17:
Hi, I really like this project, thanks very much for making this happen!

My IOTAppstory logs and serial output keep showing"Error: Max. callhome / min / device reached!"

And I have this error message in Serial: " Error: Header did not start with magic byte 0xE9"

Any help would be very much appreciated!

=======================================

My setup:

Hard:  NodeMcu Clone Amica ESP12E 4mb, no pins connected, power supply 5Vdc with 2Amp available.
Firm:  NodeMCU Version 0.9,  SDK Firmware - 2.2.0 08/062018
Core: ESP arduino communitity core - 2.4.2
IDE:   Arduino/Windows 64  1.8.5
IDE config, and debug settings:
          Board NodeMCU v0.9
          Flash size: 4M (1M SPIFFS)
          Debug Port:Disabled
          Debug Level: non
          lwlp Variant: Version 1.4 higher bandwidth
          VTables: Flash
          CPU: 80MHz
          Upload: 115200
          Erase Flash: only Sketch
          Programmer: AVRISP mkll

libraries:    IOTAppStory.h  2.0.0
                  ESP-Library-2.0.1   Latest commit 7e9ed22 on Jul 24
                  ESPAsyncTCP-master  Latest commit 7e9ed22 on Jul 24
                  ESPAsyncWebServer-master Latest commit 7e9ed22 on Jul 24

sketch:      IASBlink V1.0.2

Serial Output:

_EEPROM Configuration found
*-------------------------------------------------------------------------*
 Start INITLoader v1.2.0
*-------------------------------------------------------------------------*
 Mode select button: GPIO0
 Boardname: initloader-151
 Update on boot: 0
*-------------------------------------------------------------------------*
 rtcMem
 bootTimes since powerup: 3
 boardMode: N
*-------------------------------------------------------------------------*
 Connecting to WiFi AP
 .....
 WiFi connected

 Device MAC: 68:C6:3A:C2:D3:97
 Device IP Address: 192.168.88.244
 MDNS responder started: http://initloader-151.local

 To use mDNS Install host software:
 - For Linux, install Avahi (http://avahi.org/)
 - For Windows, install Bonjour (https://commaster.net/content/how-resolve-multicast-dns-windows)
 - For Mac OSX and iOS support is built in through Bonjour already
*-------------------------------------------------------------------------*

 Calling Home

 Checking for App(Sketch) updates from: https://iotappstory.com/ota/esp8266-v2.0.1.php
 Download & Process update...[httpUpdate] Header read fin.

[httpUpdate] Server header:

[httpUpdate]  - code: 200
[httpUpdate]  - len: 5011
[httpUpdate]  - MD5: 4821c161e617223cf5c9103e1412855d
[httpUpdate] ESP8266 info:

[httpUpdate]  - free Space: 2748416
[httpUpdate]  - current Sketch Size: 395392
[httpUpdate]  - current version: 1.2.0_

 Error: Header did not start with magic byte 0xE9
 Reboot necessary!

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

Onno-Dirkzwager, 2018-12-17:
@Cloolalang,
consider this a warning: "Error: Max. callhome / min / device reached!"
Your devices are limited to calling home once a minute. In practice you will setup your devices to something more reasonoble like once an hour or once very 24 hours.

Error: Header did not start with magic byte 0xE9
Your esp does not reconize the file as being a firmware file. Most often caused by uploading the wrong file: spiffs file instead of the firmware(app), file compiled for other platform.

I hope this helps. Otherwise we will dig deaper....

Cloolalang, 2018-12-18:
Hi Onno, 
Thanks for your quick response.
The sketch (IASBlink V1.0.2) was calling home every 60 seconds, I changed it to 70 seconds but get the same Error: Max. callhome.

In the IAS Control panel, edit APP page, I upload IASBlink.ino and when saved it gets converted into <projectname>-<version>-8420.bin, I then set it on-line and in the project, also on-line. So why would my ESP not reconize the uploaded file as firmware?

I didnt try to upload an IDE complied spiffs file, I will try it today.

Am I missing a module in the Nodemcu firmware build? (im using the default modules)  https://nodemcu-build.com/

Cheers
P
(IOTAPPSTORY account - PatrickIOT)

Cloolalang, 2018-12-18:
Hey!,
OK Im learning a few import things today:
1) Read the Wikki
2) Read the Wikki again.
3) Learn how to save, Export Compiled Binaries in the Arduino IDE
4) Dont try to load an Arduino .INO sketch into the Edit Apps, Firmware Sketch area, it has to be a compiled .bin
5) Version control, IDE filename control
6) Use Arduino IDE degbugging
7) Use the https://iotappstory.com/cp/logs rather than the dashboard logs (which are slower to update)
8) when your ESP is resetting often due to incorrect firmware upload cycling etc, you get the  "Error: Max. callhome / min / device reached!"... 
Thanks for this great project, im will upload my projects when they are good to go!

closed 2018-12-18 by Cloolalang