iotappstory/ESP-Library #96: Cant upload IOSLoader sketch

YT37, 2018-12-28:


CwlBroeders, 2018-12-28:
Hello YT37,

Could u give some information as to what problem u are running into?
Are u getting an error thru serial?
What are u trying to upload to? esp8266? esp32? what module?

YT37, 2018-12-29:
sorry for not providing it earlier.I am using esp8266.It is compiling just fine but on uploading, I am getting this error:-
{In file included from D:\Codes\ArduinoCodes\libraries\IOTAppStory-ESP\examples\IASLoader\IASLoader.ino:33:0:
D:\Codes\ArduinoCodes\libraries\IOTAppStory-ESP\src/IOTAppStory.h:33:96: fatal error: ESPAsyncWebServer.h: No such file or directory
#include <ESPAsyncWebServer.h>               // https://github.com/me-no-dev/ESPAsyncWebServer
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).}
Using these settings:-
[image]

CwlBroeders, 2018-12-29:
looks like the compiler cannot find the ESPAsyncWebServer.h file
the iotappstory library needs this file to be installed
For a full list of dependancies, please check
https://iotappstory.com/wiki/arduino-esp/how-to-get-started/prerequisites

YT37, 2018-12-29:
Yes, I have installed it a couple of times and moved it to library folder.But it still isn't uploading