iotappstory/ESP-Library #98: callHome defaults SPIFFS

Bolukan, 2019-01-02:
If manually invoked callHome, you can turn SPIFFS off:
```C
IAS.callHome(false);
```
https://github.com/iotappstory/ESP-Library/blob/f8b0da1f922f6c787cce6a1d140f76b15ad464be/src/IOTAppStory.cpp#L611

but the bool will default to be true, if called through the timer, after preSetAutoUpdate(true) or the button press. I would expect this to be configurable too?

https://github.com/iotappstory/ESP-Library/blob/f8b0da1f922f6c787cce6a1d140f76b15ad464be/src/IOTAppStory.cpp#L1142 https://github.com/iotappstory/ESP-Library/blob/f8b0da1f922f6c787cce6a1d140f76b15ad464be/src/IOTAppStory.cpp#L236 https://github.com/iotappstory/ESP-Library/blob/f8b0da1f922f6c787cce6a1d140f76b15ad464be/src/IOTAppStory.cpp#L1226

Onno-Dirkzwager, 2019-01-02:
Good point,
what would be your prefered solution?
- preSetUpdateSpiffs(bool)
- define like with the nextion
- other:

Bolukan, 2019-01-02:
Just wonder where you define U_SPIFFS and U_NEXTION? I love it.

But to the point: I would prefer a #define OTA_SPIFFS. That said, it would be quite fixed choice for an application (changes between versions is possible, but little bit complicated). 
preSetUpdateSpiffs(bool) makes it facultative, but is that desirable? One of my thoughts is the EEPROM definition changes with al those #defines, so EEPROM is not exchangable between Apps? As long it is managed #define works ok.
I don't believe the amount of memory used (ROM/code) is that important. Most of the code is reused. The extra amount of RAM used should be very limited.

Onno-Dirkzwager, 2019-01-02:
The 8266 sets U_FLASH and U_SPIFFS in the core(update.h) we added U_NEXTION
https://github.com/iotappstory/ESP-Library/blob/develop/src/espressif/esp8266/otaUpdate.h

The 32 does not have these(yet...) so we add all 3
https://github.com/iotappstory/ESP-Library/blob/develop/src/espressif/esp32/otaUpdate.h

These are all for internal use. The defines we provide for our library are set in the config.h as you already know.

The upcomming 2.1.0 will have a few more config.h defines one of them:
#define NEXT_OTA	true	// Do you want to OTA update your Nextion display? | true / false
https://github.com/iotappstory/ESP-Library/blob/develop/src/config.h#L29

Ill set something simelar up for SPIFFS

Onno-Dirkzwager, 2019-02-11:
https://github.com/iotappstory/ESP-Library/blob/develop/src/config.h#L28

Onno-Dirkzwager, 2019-05-05:
Im closing this issue as this is now an option in the latest release!

closed 2019-05-05 by Onno-Dirkzwager