iotappstory/ESP-Library #101: Compatability with Wemos (Lolin) OLED shield v2.1.0

877dev, 2019-01-27:
Hi,
First thanks to Christiaan Broeders for the help so far. I have run into the following issue when using Lolin OLED shield. 

I am not using a button shield, as the OLED has buttons on it already (although they maybe need a LOLIN_I2C_BUTTON.h library to run).

This is the shield, as you can see it already has buttons on it, and is 64 x 48
https://wiki.wemos.cc/products:d1_mini_shields:oled_shield

The OLED library I have on another project which works correctly:
https://github.com/theiothing/Adafruit_SSD1306-esp8266-64x48

This is the code used (commented out the chipId part as suggested:
https://codeshare.io/aV4VRR

RESULTS:

Total blank black screen on OLED, nothing shows.

Serial monitor displays the following:
15:37:35.958 -> 
15:37:35.958 -> 
15:37:35.958 -> ⸮⸮2⸮
15:37:36.395 ->  ets Jan  8 2013,rst cause:2, boot mode:(1,6)
15:37:36.395 -> 
15:37:44.733 -> 
15:37:44.733 ->  ets Jan  8 2013,rst cause:4, boot mode:(1,6)
15:37:44.733 -> 
15:37:44.733 -> wdt reset

Onno-Dirkzwager, 2019-01-27:
Hi @877dev,

We developed the wemos example sketches for the 1.1.0 version oled shield. This did not have the extra buttons. The oled uses the same pins, Driver IC: SSD1306 and IIC Address: 0x3C or 0x3D. So I would expect this to work.

The buttons however will not work as they are I2C and the IOTAppStory library only alows you to use the gpio pins for buttons.

We did not realize these changes in the new shields and will update our examples in future releases.

Which sketch / app are you trying to use? And as mentioned above you will need a button. (wire to ground for button press will do)

877dev, 2019-01-27:
Update:
Had a go changing the code and there's so many errors, I gave up.

Onno-Dirkzwager, 2019-01-27:
As mentioned above we never made these examples for this shield. And it will take some time before its in the next release. Our apologies for this inconvenience.

Untill then just coninue using the normal IASLoader to add your device to our servers. You can still use your oled shield in your own projects with the libraries that work for you! 
Its just not compatible with our examples. yet....

Ill leave this issue open as a reminder untill we updated the examples.

877dev, 2019-01-28:
@Onno-Dirkzwager I did not realise I could use the normal IASLoader, thank you.

PS - please let me know if you want anything testing in the meantime..

CwlBroeders, 2020-08-03:
@877dev The wemos example sketches have been updated to work with the v2.1.0 OLED shield

closed 2020-08-13 by CwlBroeders