iotappstory/ESP-Library #102: Values with double quotes in text areas not working properly

abaskin, 2019-02-11:
When I enter a value that includes double quotes into a text area field, the value is accepted and saved properly. However when I later enter configuration mode and select the setting tab the resulting page is blank. An example value is below. I see the same result for both one line and multi line values that include double quotes.

`{
	"name": "Test Module 1",
	"switch": [{
		"name": "Test Switch 1",
		"pin": "D1",
		"led": ["D4"]
	}]
}`

Onno-Dirkzwager, 2019-02-11:
Hi @abaskin,
just to be sure: 
This is the entire value you are trying to store

> """"{ "name": "Test Module 1", "switch": [{ "name": "Test Switch 1", "pin": "D1", "led": ["D4"] }] }

abaskin, 2019-02-11:
Yes, this is the entire value I am trying to store. The reason for using JSON as opposed to individual fields is that the app supports a number of different things attached to the device. So one device may have a DHT sensor attached and another may have a switch, relay and led attached. Supporting each attached thing with separate fields would make the setup page unwieldy at best especially for something like LED's where there may be more than one attached to the device.

Onno-Dirkzwager, 2019-02-11:
@abaskin 
I see. 
We will dive into the code. I think we should add js script that escapes the quotes with backslashes before sending the string to your esp for saving.

Maby you can help by testing if this works by just storing this value into a var without using the IAS addfield stuff? Storing should not be the problem. Im wondering if the json still gets read correctly when taken from this var.

Onno-Dirkzwager, 2019-03-01:
@abaskin FIXED!
It turns out saving your values was not the problem. But it did break the json format that gets sent to your browser when running the config pages. So after trying to do the usual backslahes we decided to urlencode the value before sending it to the browser. Js scripts in the config pages handel decoding.

**Example:**
[
    {
        "l":"Textarea", 
        "v":"{ %22name%22: %22Test Module 1%22, %22switch%22: [{ %22name%22: %22Test Switch 1%22, %22pin%22: %22D1%22, %22led%22: [%22D4%22] }] }", 
        "n":"1", 
        "m":"121", 
        "t":"T"
    }
]

Changes were made here: https://github.com/iotappstory/ESP-Library/commit/c42ab038fb6d76af4e3bb1410ff5732d8fbd1e3c

We need a few people to test this before we can send out a new release. Maby you can help?

Easiest thing would be te replace your IOTAppStory.h and .cpp files with the changed ones from the master.

Onno-Dirkzwager, 2019-03-02:
@abaskin @CwlBroeders added your [suggested](https://github.com/iotappstory/ESP-Library/commit/c42ab038fb6d76af4e3bb1410ff5732d8fbd1e3c#commitcomment-32557890) changes: 
https://github.com/iotappstory/ESP-Library/blob/master/src/IOTAppStory.cpp#L1492

Onno-Dirkzwager, 2019-03-03:
Added a new release: https://github.com/iotappstory/ESP-Library/releases/latest

Should be in the library manager within an hour.

Onno-Dirkzwager, 2019-03-04:
@abaskin I think we can close this issue do you agree?

abaskin, 2019-03-05:
Yes

closed 2019-03-05 by abaskin