iotappstory/ESP-Library #103: IOTAppStory::addField() crashes with heap variables

tbaumann, 2019-02-19:
I have the feeling addField() somehow gets all confused if I use heap variables.

```c
#define SENSIDX 2 // Number of sensors on bus - 1.
char* sensorNames[SENSIDX];
char* sensorLabels[SENSIDX];

  for(int index = 0; index <= SENSIDX; index++){
    sensorLabels[index] = (char*) malloc(15);
    sensorNames[index]  = (char*) malloc(16 + 1); // addField length parameter seems to be strlen not bytes

    sprintf(sensorLabels[index], "Sensor %d Name", index);
    strcpy(sensorNames[index], "Unnamed");

    IAS.addField(sensorNames[index], sensorLabels[index], 16, 'L');
  }
```

I suspect that the `const` for `label` is meant seriously?

Apart from the fact that I deeply dislike that soution. It also doesn't work. (Guru meditation)

```c
const char* sensorLabels[] = {"Sensor 0", "Sensor 1", "Sensor 2"} ;
```
No malocs and strcpy this time. 

I'm no expert. But that looks const enough to me.

Is `const char *fieldLabel` indeed the problem here?

Can I perhaps mark it const and promise I won't change the value at runtime but initualise it dynamically?

tbaumann, 2019-02-19:
```c
  IAS.addField(sensorNames[0], "Sensor 0", 16, 'L');
  IAS.addField(sensorNames[1], "Sensor 1", 16, 'L');
  IAS.addField(sensorNames[2], "Sensor 2", 16, 'L');
```
Crashes equally. Const-ness of the label isn't even the problem here.

I'm quite confused now.

Onno-Dirkzwager, 2019-02-26:
Hi @tbaumann,
you are using this function in a way it was never intended for. But I get where you are comming from... as a programmer you do not want to repeat things... and rather setup a loop for this. We will definitely look into this.

For now you will have to use it like in the [examples](https://github.com/iotappstory/ESP-Library/blob/master/examples/IASBlink/IASBlink.ino#L56):
  char* LEDpin    = "2";
  char* blinkTime = "1000";

  IAS.addField(LEDpin, "ledpin", 2, 'P');
  IAS.addField(blinkTime, "Blinktime(mS)", 5, 'N'); 

So in your case:
  char* sensorName0 = "Example 1";
  char* sensorName1 = "Example 2";
  char* sensorName2 = "Example 3";

  IAS.addField(sensorName0, "Sensor 0", 16, 'L');
  IAS.addField(sensorName1, "Sensor 1", 16, 'L');
  IAS.addField(sensorName2, "Sensor 2", 16, 'L');

(sensor name value,      html field name,          max char that can be saved from html,        fieldtype)

tbaumann, 2019-02-26:
Thanks Onno, I can certainly do it that way. No problem. The code deletion was in the end not really too bad.
Perhaps I do it one day in preprocessor if I can actually be bothered. 

Good to know I didn't just do something trivially wrong, thanks. I don't so C very often. 😄

Onno-Dirkzwager, 2019-02-26:
Great, 
good luck with your projects! And if your question is answered please close this issue. (it's better for the stats if the issue opener closes it)

closed 2019-02-27 by tbaumann