iotappstory/ESP-Library #105: Documentation Suggestion - IAS.onFirstBoot and WiFi.onEvent

QuantusHoka, 2019-02-26:
I see that there is an undocumented callback for IOTAppStory::onFirstBoot(THandlerFunction value). I am setting-up some EEPROM storage and find this callback very useful, and others may as well. I suggest adding it to the wiki.

Similarly, IOTAppStory uses the WiFi library, but does not appear to use WiFi.onEvent; please correct me if I am wrong. It might, perhaps, be useful to note that the callback is unused, and thus available for use in our applications.

Finally, I'm new to (and thus still learning) the github thing, so please let me know if there is a more appropriate way to make such suggestions.

Onno-Dirkzwager, 2019-02-26:
Hi @QuantusHoka,
- onFirstBoot you are right we will make a note of this. (we use it for EEPROM internally as well)
- WiFi.onEvent same! Coincidentally we had the same suggestion on discord.
- We try to use github issues for primarily for technical issues, bugs, feature suggestions etc. But this was a great first issue!

Onno-Dirkzwager, 2019-03-01:
@QuantusHoka 
I updated the Wiki: https://iotappstory.com/wiki/arduino-esp/library-api-20/api-methods/callbacks()

The onFirstBoot is on top and the Wifi.onEvent on the bottom. We will setup a dedicated Wifi page in a future update. But for now I think your suggestions have been addressed.

If you agree please close this issue!

QuantusHoka, 2019-03-02:
Excellent. Thanks.

closed 2019-03-02 by QuantusHoka