iotappstory/ESP-Library #106: IASLoader - login fails - if WiFi ssid contains "-"

KlausJuhantalo, 2019-02-27:
IASLoader - login fails - if WiFi ssid contains "-", even the WiFi credentials are correct.

After i changed WiFi ssid name to name that didnt exist "-" character it worked.

Used esp32 (TTGO LoRa32-OLED-V1).

southseaboy, 2019-02-27:
Use of some special characters in the password entered into the IASloader config webpage also causes a connect failure - & and % seem troublesome, whereas ^ and # seem to be OK.

Onno-Dirkzwager, 2019-02-27:
As southseaboy pointed out on Discord. These special char do work when hardcoded with the presets. 
So this narrows it down to esp <> browser communication.

Onno-Dirkzwager, 2019-02-27:
@KlausJuhantalo @southseaboy 
Testers wanted!
I made a change to the [master](https://github.com/iotappstory/ESP-Library)

This makes most special characters acceptable by your ESP.
@ # * - = ( ) _ $ / ; : ' ! , . "

Still not ok:
& % +

% and + are caused by how the ESPAsyncWebServer [urlDecode](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebRequest.cpp#L954) method works.

& needs more research but seems te be used by the esp instead of printing it as a char

Bolukan, 2019-02-28:
So we need also an espprintencoder. Anybody knows of one?

southseaboy, 2019-02-28:
I know 0 from web coding, but I can enter my password which contains some of the still offending characters into Tasmota's WiFi config page and it works.  In Tasmota the password is not shown in clear on the screen.

southseaboy, 2019-02-28:
Testing here confirms expected behaviour listed by @Onno-Dirkzwager  above with one exception:
**%**           .... works for me, 
**& +**        ... do not work

Onno-Dirkzwager, 2019-02-28:
@southseaboy 
Where in the line did you put the %?
If I put it on the end of the line it works. Halfway the line will take down the next char. (which is not that strange thinking about urlencoding)

southseaboy, 2019-02-28:
Ah yes - I was just appending the special character(s) at then end of some normal text

Onno-Dirkzwager, 2019-02-28:
Warning! It seems the latest fix for the spec char breaks device registration.

Onno-Dirkzwager, 2019-03-03:
Added a new release: https://github.com/iotappstory/ESP-Library/releases/latest

Should be in the library manager within an hour.

Onno-Dirkzwager, 2019-03-04:
We will make a note for the following characters & % + and close this one as the original issue has been solved. Feel free to reopen if deemed necessary.

closed 2019-03-04 by Onno-Dirkzwager

DocDoo, 2019-08-15:
Hi @Onno-Dirkzwager 

On the 4th of March you mentioned

> We will make a note for the following characters & % + and close this one as the original issue has been solved. Feel free to reopen if deemed necessary.

Where was this note made? As I would like to review what you wrote as I suspect that I'm facing the same problem but with the password for my WiFi and not the SSID for it. In my current WiFi password I have 19 chars of lower and uppercase chars and numbers and the special char &.