iotappstory/ESP-Library #107: Problems registering/activating new device today 28/2/19

Cloolalang, 2019-02-28:
Hey.
Firstly, Thank-you, this is a fantastic project, Its helping me a great deal in my IOT dev.

Platform/environment:
NodeMCU 0,9 12E
Core - 2.4.2 (Int), Arduino- 2.4.1
IDE:Arduino/Windows 64  1.8.5
IOTAPPSTORY: 2.0.0
INITLoader V1.2.0

I have a few devices all registered without issues, today no, I am having no luck at all!

Each time I fill in a device name and board type and Next, nothing happens :/
I can access the wifi scan pages ok, so I know my device is connected to the wifi and I have checked that the wifi is online.

Tried already:
1) Erased and re-flashed, ESP and uploaded un-edited INITLoader V1.2.0
2)  Web config page, (re) register device   different Windows browsers, various names and NodeMCU types
3) Web config on Android devices as per 2

What am I doing wrong?

Patrick

Onno-Dirkzwager, 2019-02-28:
@Cloolalang nothing sir! We are working on a fix as we type.

Onno-Dirkzwager, 2019-02-28:
@Cloolalang fixed!

Onno-Dirkzwager, 2019-02-28:
@Cloolalang It would be wise to start over and reupload the Iasloader app. As previous attempts might have corrupted the eeprom settings.

Cloolalang, 2019-03-01:
Thank-you Onno!, it works.

closed 2019-03-01 by Cloolalang