iotappstory/ESP-Library #110: project name and app type

moijerem59, 2019-03-22:
Hello, and it possible to define the characteristic directly in the code arduino without passes by the web page? for
project name and app type
?
and it possible to increase the limitation of the 20 devices?
Thank you for your work. Your solution and just magic

Onno-Dirkzwager, 2019-03-22:
Hi @moijerem59,
Project name and app are IAS server(CP) settings and cannot be set from the sketch. You(your browser) can only do this in the CP and Config pages when you are logged in.

But what would you like to accomplish? Because we have a api and special sketch comming that might serve your needs.

About the device limit. 
If you are a company we have some beta pro accounts running. We can talk about the details in private on Discord or over skype.
And we have some paid options comming for the main IOTAppStory.com website that will let you extend your account.

How many do you need?
I can set you up for a free trial for more. Or if you have a cool project worthwhile sponsering I can help you out right now.

moijerem59, 2019-03-22:
Hello, thank you for your return.
We have developed an equipment that uses the wifi manager library. Actually we use the ota via the web page. Your solution attracts us for the rest of our development.
But for reasons of ergonomics the fact of having to connect each equipment to our network to configure the project and to apply it to this equipment and a brake d or our question.
If the connection between the device and the project on iotappstore could be directly edited via the code. That would make it easier for us.
Regarding the limitation of the number of devices at the moment this is not a brake for our test. If your solution satisfies us, we will have to discuss the possibilities to increase the number.
Sorry for my English.

Onno-Dirkzwager, 2019-03-22:
@moijerem59 
So what I beleave you need is actually comming! 

- You setup a project in the CP: Name & app. Save
- Copy the created project id.
- Past this to the special Initial loader sketch together with wifi cred, thingerprint, activation code.
- Compile & Upload
- Watch your device get added to your project and imidiatly download the lastest app.

This procedure supports adding large batches of devices!

moijerem59, 2019-03-22:
How should we proceed ? did you have an example for us to start trying our side?

Onno-Dirkzwager, 2019-03-22:
@moijerem59 it's comming! As in not released yet!
I think we should procede on others channels as this is not an library issue.
Could you come to Discord or Skype?

moijerem59, 2019-03-22:
yes of course, here is my discord
PRX # 8012

Onno-Dirkzwager, 2019-03-22:
Found you on Discord. Closing.

closed 2019-03-22 by Onno-Dirkzwager