iotappstory/ESP-Library #116: IOTAPPStory.com timeout

Cloolalang, 2019-04-08:
Hi, I raised this issue earlier today and closed it thinking it as solved.
Bur no.
The site: https://iotappstory.com/cp times-out here for me in South Africa (Mweb)

Cloolalang, 2019-04-08:
From Cape Town at 16:49 SAT
Tracing route to iotappstory.com [37.46.193.199]
over a maximum of 30 hops:

  1     1 ms    <1 ms    <1 ms  192.168.88.1
  2     3 ms     3 ms     3 ms  197-84-0-74.cpt.mweb.co.za [197.84.0.74]
  3     7 ms     3 ms     3 ms  197-84-97-130.cpt.mweb.co.za [197.84.97.130]
  4   146 ms   145 ms   145 ms  te0-7-0-3.cpt-p-2.optinet.net [197.84.7.2]
  5   145 ms   144 ms   145 ms  197-84-4-215.cpt.mweb.co.za [197.84.4.215]
  6   147 ms   145 ms   145 ms  be-2-778-thd-up-2.optinet.net [176.67.177.148]
  7   152 ms   147 ms   144 ms  be-16-778-thd-linx-1.optinet.net [176.67.177.149]
  8   151 ms   151 ms   151 ms  195.66.226.224
  9   165 ms   182 ms   154 ms  37.46.195.29
 10   221 ms   158 ms   212 ms  37.46.195.40
 11     *        *        *     Request timed out.
 12     *        *        *     Request timed out.
 13     *        *        *     Request timed out.
 14     *        *        *     Request timed out.
 15     *        *        *     Request timed out.
 16     *        *        *     Request timed out.
 17     *        *        *     Request timed out.

Onno-Dirkzwager, 2019-04-08:
@Cloolalang,
thanks for the info I just shared it with the cloud / hosting provider.

Onno-Dirkzwager, 2019-04-08:
@Cloolalang has your situation improved? 
Because for me and my tests it has. 

I also saw your route trace was from saterday (From Cape Town at 16:49 SAT)
Which corresponds with the service being offline as mentioned in the previous issue.

Pingen naar 37.46.193.199 met 32 bytes aan gegevens:
Antwoord van 37.46.193.199: bytes=32 tijd=24 ms TTL=53

Hop | Node | Response Time(ms)
1 | Internal (Internal) | 1.144
2 | bgp-rt02-gw.jhb.domains.co.za (169.239.216.242) | 1.153
3 | 181-64-148-197.as37497.za.net (197.148.64.181) | 2.061
4 | 42-71-148-197.as37497.za.net (197.148.71.42) | 163.662
5 | 162-68-148-197.as37497.za.net (197.148.68.162) | 163.649
6 | 14-71-148-197.as37497.za.net (197.148.71.14) | 161.742
7 | 33-66-148-197.as37497.za.net (197.148.66.33) | 162.060
8 | 195.66.226.224 (195.66.226.224) | 168.930
9 | 37.46.195.29 (37.46.195.29) | 185.478
10 | 37.46.195.40 (37.46.195.40) | 194.405
11 | iotappstory.com (37.46.193.199) | 173.094

Cloolalang, 2019-04-08:
Hey Onno, 
I just tried it now 22.45 8/4/19 and now it works
My trace was from today South African Time (SAT)
:)
here it is now:
C:\Users\User>tracert www.iotappstory.com

Tracing route to iotappstory.com [37.46.193.199]
over a maximum of 30 hops:

  1    <1 ms    <1 ms    <1 ms  192.168.88.1
  2     3 ms     3 ms     3 ms  197-84-0-74.cpt.mweb.co.za [197.84.0.74]
  3     5 ms     3 ms     4 ms  197-84-97-130.cpt.mweb.co.za [197.84.97.130]
  4   149 ms   146 ms   149 ms  te0-7-0-3.cpt-p-2.optinet.net [197.84.7.2]
  5   145 ms   144 ms   145 ms  197-84-4-215.cpt.mweb.co.za [197.84.4.215]
  6   148 ms   148 ms   145 ms  be-2-778-thd-up-2.optinet.net [176.67.177.148]
  7   149 ms   145 ms   149 ms  be-16-778-thd-linx-1.optinet.net [176.67.177.149]
  8   151 ms   151 ms   151 ms  195.66.226.224
  9   165 ms   164 ms   168 ms  37.46.195.29
 10   161 ms   155 ms   183 ms  37.46.195.40
 11   158 ms   158 ms   153 ms  iotappstory.com [37.46.193.199]

Trace complete.

Onno-Dirkzwager, 2019-04-08:
@Cloolalang 
Thanks for your initial report and thanks for testing and providing the extra input.

Cloolalang, 2019-04-08:
Welcome!
Thanks for this very useful gift to our community!

Onno-Dirkzwager, 2019-04-08:
@Cloolalang 
Appreciated! 
Btw. we have lots of exciting updates coming! So if you like it now, things will only get beter.

Cloolalang, 2019-04-16:
Hi!
Same isue today from Cape Town :(

========================================
Pinging iotappstory.com [37.46.193.199] with 32 bytes of data:
Request timed out.
Request timed out.
Request timed out.
Request timed out.

Ping statistics for 37.46.193.199:
    Packets: Sent = 4, Received = 0, Lost = 4 (100% loss),
=========================================
acing route to iotappstory.com [37.46.193.199]
over a maximum of 30 hops:

  1    <1 ms    <1 ms    <1 ms  192.168.88.1
  2     3 ms     3 ms     2 ms  197-84-0-74.cpt.mweb.co.za [197.84.0.74]
  3     4 ms     3 ms     3 ms  197-84-97-130.cpt.mweb.co.za [197.84.97.130]
  4   148 ms   145 ms   145 ms  te0-0-0-1.cpt-p-2.optinet.net [197.84.7.34]
  5   145 ms   145 ms   145 ms  197-84-4-215.cpt.mweb.co.za [197.84.4.215]
  6   146 ms   145 ms   145 ms  be-2-778-thd-up-2.optinet.net [176.67.177.148]
  7   186 ms   145 ms   147 ms  be-16-778-thd-linx-1.optinet.net [176.67.177.149]
  8   151 ms   151 ms   151 ms  195.66.226.224
  9   158 ms   164 ms   163 ms  37.46.195.29
 10   173 ms   352 ms   300 ms  37.46.195.40
 11     *        *        *     Request timed out.
 12     *        *        *     Request timed out.
 13     *        *        *     Request timed out.
 14     *        *        *     Request timed out.
 15     *        *        *     Request timed out.
 16     *

Onno-Dirkzwager, 2019-04-16:
Hi @Cloolalang sorry to hear...again!
I passed this to our cloud provider who is researching this as I type. And did multiple worldwide speedtests myself. All countrys / citys came out reasonable fast except for tokio & shanghai.

You are clearly having problems but as the initial tests show this is going to be a tricky one.

Any extra tests, ideas & thoughts from your side are welcome!
Ill consider upgrading your account as soon as we pinpointed the cause and are able to fix it for good.

Cloolalang, 2019-04-16:
Hey, Thanks for your reply
So it just started working again:
C:\Users\User>ping www.iotappstory.com

Pinging iotappstory.com [37.46.193.199] with 32 bytes of data:
Reply from 37.46.193.199: bytes=32 time=161ms TTL=55
Reply from 37.46.193.199: bytes=32 time=155ms TTL=55
Reply from 37.46.193.199: bytes=32 time=157ms TTL=55
Reply from 37.46.193.199: bytes=32 time=152ms TTL=55

Ping statistics for 37.46.193.199:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 152ms, Maximum = 161ms, Average = 156ms

C:\Users\User>tracert 37.46.193.199

Tracing route to iotappstory.com [37.46.193.199]
over a maximum of 30 hops:

  1    <1 ms    <1 ms    <1 ms  192.168.88.1
  2     2 ms     3 ms     2 ms  197-84-0-74.cpt.mweb.co.za [197.84.0.74]
  3     6 ms     3 ms     3 ms  197-84-97-130.cpt.mweb.co.za [197.84.97.130]
  4   146 ms   144 ms   145 ms  te0-0-0-1.cpt-p-2.optinet.net [197.84.7.34]
  5   147 ms   145 ms   145 ms  197-84-4-215.cpt.mweb.co.za [197.84.4.215]
  6   146 ms   145 ms   145 ms  be-2-778-thd-up-2.optinet.net [176.67.177.148]
  7   147 ms   170 ms   146 ms  be-16-778-thd-linx-1.optinet.net [176.67.177.149]
  8   150 ms   151 ms   176 ms  195.66.226.224
  9   165 ms   167 ms   205 ms  37.46.195.29
 10   220 ms   196 ms   206 ms  37.46.195.40
 11   153 ms   153 ms   153 ms  iotappstory.com [37.46.193.199]

Trace complete.

C:\Users\User>

Onno-Dirkzwager, 2019-04-16:
@Cloolalang thanks for the feedback. 
Now we have to find out what happend. Compairing your traces it looks like you get blocked in the Dutch datacenter. Ill pass this on to our provider.

Onno-Dirkzwager, 2019-04-22:
@Cloolalang how has your connection been last week?

Onno-Dirkzwager, 2019-05-05:
Closing this issue. Feel free to reopen if deemed necessary!

closed 2019-05-05 by Onno-Dirkzwager

Cloolalang, 2019-06-13:
Hey Onno, 
I have started to get issue with my OTA again here in South Africa.
Has something changed? 

I see in the logs page "receiving", but my device does not reset
Cheers
Patrick

My ESP8266 serial dump:

**pm open,type:2 0
[HTTP-Client][returnError] error(-11): read Timeout
[HTTP-Client][returnError] tcp stop
 Failed. Lost Wifi, Server down OR HTTPS problems
[HTTP-Client][end] tcp is closed

 Checking for SPIFFS updates from: https://iotappstory.com/ota/esp8266-v2.0.1.php
[HTTP-Client][begin] url: https://iotappstory.com/ota/esp8266-v2.0.1.php
[HTTP-Client][begin] host: iotappstory.com port: 443 url: /ota/esp8266-v2.0.1.php
[HTTP-Client][begin] httpsFingerprint: 34:6D:0A:26:F0:40:3A:0A:1B:F1:CA:8E:C8:0C:F5:14:21:83:7C:B1
[HTTP-Client] connected to iotappstory.com:443
[HTTP-Client] sending request header
-----
GET /ota/esp8266-v2.0.1.php HTTP/1.0
Host: iotappstory.com
User-Agent: ESP-http-Update
Connection: close**