iotappstory/ESP-Library #121: library to v. 2.1.0 RC 1.1 - No call home

ivanovlk, 2019-06-02:
On one of my projects i updated library to v. 2.1.0 RC 1.1.
Suddenly after update (via IOTAppStory). 
My device stopped calling home. Does not appear in dashboard or Log.

This is probably related to: 

 IAS.setCallHome(true);
and
void IOTAppStory::setCallHome(bool callHome) {} // <----- deprecated left for compatibility. Remove with version 3.0.0

PS: Even restarting the device does not force it to call home!?
PS1: Another strange behavior is that Wi-Fi re-connect on disconnect does not work anymore ?!

Onno-Dirkzwager, 2019-06-02:
Hi @ivanovlk ,
does your device not even attempt to call home? Or does it try and end in some sort of an error?

ivanovlk, 2019-06-02:
Hi,

Yes. Error in Console :(
Seems some kind of cert issue ?!

>  Checking for App(Sketch) updates from: https://iotappstory.com/ota/updates.php
 Could not load Certificate!

 Checking for SPIFFS updates from: https://iotappstory.com/ota/updates.php
 Could not load Certificate!

-

Onno-Dirkzwager, 2019-06-02:
@ivanovlk 
This is the release candidate for 2.1.0 and de documentation is still beging worked on for the final version.

Previous versions for the esp32 used a hardcoded certificate for tls connections. In this upcomming version you can upload and maintain certificates in config mode.

So please go into config mode, cetificates and click on the get ias cert button or upload your own.

Onno-Dirkzwager, 2019-06-02:
@ivanovlk 
Just to let you know:
I found a bug regarding SPIFFS / certificates on new devices.... I think I previously tested to much on the same device....
Ill research this and get back to you

ivanovlk, 2019-06-03:
@Onno-Dirkzwager Thanks!

BTW i managed to make it work.

As you said i entered config and downloaded Cert. Unfortunately this was not enough :)
I had to delete device from dashboard and re-create it.
Now everything works as expected :)

BTW few ideas:
1) Is it possible to flash new firmware via Config page ?
2) Is there a function which can be triggered by MQTT for example which will force reboot the device in Config mode?

Onno-Dirkzwager, 2019-06-03:
Thanks for the feedback. 
Deleting & recreating should not have been necessary. Simply going to the IAS page in config should have reconized your device and reregistered it. Keeping al previous project settings.

1. Checkout the Other Examples/ [OfflineSpiffsConfigPages/](https://github.com/iotappstory/ESP-Library/tree/master/examples/Other%20Examples/OfflineSpiffsConfigPages) and make sure to set the defines as suggested on line 28.
2. We don't have any MQTT functions. But you can build something yourself and use IAS.espRestart(char mmode)    N = normal, C = config mode.

Onno-Dirkzwager, 2019-06-04:
@ivanovlk 
- Did my previous answers help?
- I just made a PR #122 which fixes the SPIFFS bug I mentioned
- Can we close this issue?

ivanovlk, 2019-06-05:
Hi,

@Onno-Dirkzwager Thanks!

Yes. I already used them both :) Works like a charm :)

Thanks for everything :)

BR,

closed 2019-06-05 by ivanovlk