iotappstory/ESP-Library #127: Dump and boot loop with new RC2

ivanovlk, 2019-07-01:
Hi @Onno-Dirkzwager,

There seems to be a dump and re-boot loop with RC2.

I made a test sketch on new device before i start using RC2 library.

So brand new device, brand new app.
I just used code snippets from my previous projects.

After flash -> Boot loop.

Dump seems to be caused by IAS.presetWifi(SSID, PASS) method.

Inside it i see EEPROM reference which in my opinion causes the dump because something is missing?! Data partition or something ??

If i remove the presetWifi everything works but no internet connection is available and device goes in config mode after few re-tries. This is intended as far i know :)

Didn't have time to check if it works when i put SSID and pass in config mode. I guess it will work.

BR,

ivanovlk, 2019-07-21:
One more side effect i noticed !?

WiFi does not get re-connected when connection is lost (e.g. router reboot or something).

Strange thing is i tried to WiFi.reconnect();

But it didn't work too?!

BR,

Onno-Dirkzwager, 2020-12-21:
@ivanovlk sorry for the late reply, we finally found out what it was.... small error when switching background code and never updated this function. I could never replicate the error.... but discovered it did hapen on devices that I used this function with for the second time....

So it should be fixed in the RC4 release. https://github.com/iotappstory/ESP-Library/releases/tag/2.1.0-RC4

Feel free to reopen this issue if deemed necessary.

closed 2020-12-21 by Onno-Dirkzwager