iotappstory/ESP-Library #129: 2.1.0 Dependency on ESPNexUpload Library

X-Illuminati, 2019-08-11:
When compiling the IASLoader example, I get the following error message:
```
In file included from /home/cbaker/Arduino/libraries/IOTAppStory-ESP/src/espressif/updateNextion.cpp:2:0:
/home/cbaker/Arduino/libraries/IOTAppStory-ESP/src/espressif/updateNextion.h:6:27: fatal error: ESPNexUpload.h: No such file or directory
  #include <ESPNexUpload.h>
```

This occurs in 2.1.0-RC1 and later. It does not occur in 2.0.2.
The ESPNexUpload library is not mentioned as one of the prerequisites.

Full compilation log for reference:
[ESP-Library-2.1.0-RC2-1-g88f0289.txt](https://github.com/iotappstory/ESP-Library/files/3490518/ESP-Library-2.1.0-RC2-1-g88f0289.txt)

X-Illuminati, 2019-08-11:
Well, after installing the ESPNexUpload library, I get several compilation errors so I probably just stick with the stable version.
```
/home/cbaker/Arduino/libraries/IOTAppStory-ESP/src/espressif/configServer.cpp: In lambda function:
/home/cbaker/Arduino/libraries/IOTAppStory-ESP/src/espressif/configServer.cpp:123:5: error: no matching function for call to 'configServer::hdlReturn(AsyncWebServerRequest*&, bool)'
     );
```

Detailed Log:
[ESP-Library-2.1.0-RC2-1-g88f0289 additional compilation errors.txt](https://github.com/iotappstory/ESP-Library/files/3490523/ESP-Library-2.1.0-RC2-1-g88f0289.additional.compilation.errors.txt)

Onno-Dirkzwager, 2020-05-03:
Hi @X-Illuminati,
sorry for letting you hanging! I made a pull request with a fix to the master. Any chance you could test these changes?

I hope to release "Release Candidate 3" this weekend.

X-Illuminati, 2020-05-09:
@Onno-Dirkzwager,

Pull #143 definitely seems to work and I don't get the error for ESPNexUpload.h anymore.

I tested with v2.6.3 of the Arduino ESP8266 toolchain. I'm sure this is a different version than I was originally using and it is also not the latest.

closed 2020-05-09 by X-Illuminati