iotappstory/ESP-Library #130: Downloading fails with "ERROR: file header verification failed!" - Content-Length: 0

cybermat, 2019-09-20:
In the last two days I've been unable to download new sketches. 
The error seems similar to one that  was reported and fixed last year at https://github.com/iotappstory/ESP-Library/issues/55 

I have recently installed the latest stable release library - IOTAppStory ESP Library 2.0.2

17:12:43.114 ->  Calling Home
17:12:43.114 -> 
17:12:43.114 ->  Checking for App(Sketch) updates from: http://iotappstory.com/ota/updates.php
17:12:44.133 -> 
17:12:44.133 ->  Received http header
17:12:44.133 ->  ---------------------
17:12:44.133 ->  HTTP/1.1 200 OK
17:12:44.133 ->  Downloading... Date: Fri, 20 Sep 2019 07:12:43 GMT
17:12:44.133 ->  Server: Apache
17:12:44.133 ->  Connection: close
17:12:44.133 ->  Transfer-Encoding: chunked
17:12:44.133 ->  Content-Type: text/html; charset=UTF-8
17:12:44.166 ->  
17:12:44.166 ->  Extracted from header
17:12:44.166 ->  ---------------------
17:12:44.166 ->  Content-Length: 0
17:12:44.166 ->  x-name: 
17:12:44.166 ->  x-ver: 
17:12:44.166 ->  x-md5: 
17:12:44.166 ->  ---------------------
17:12:44.166 -> 
17:12:44.166 ->   ERROR: file header verification failed!

CwlBroeders, 2019-09-20:
This also happens when the file itself has no contents.
(i uploaded an empty spiffs file where the .ino should go and got the same result)

have u tried reuploading your sketch?

cybermat, 2019-09-21:
Uploaded again.
The IASLogs screen says the sketch size as 407696 on download attempts - both before and after the latest upload. 
Can you see the iotappstory.com server log to  check if there is any information there? 

This new error from the esp8266 serial debug output on several download attempts isn't very helpful. 

17:32:33.072 ->  Calling Home
17:32:33.072 -> 
17:32:33.072 ->  Checking for App(Sketch) updates from: https://iotappstory.com/ota/esp8266-v2.0.1.php
17:32:44.470 -> <!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
17:32:44.470 -> <html><head>
17:32:44.470 -> <title>403 Forbidden</title>
17:32:44.470 -> </head><body>
17:32:44.470 -> <h1>Forbidden</h1>
17:32:44.470 -> <p>You don't have permission to access this resource.</p>
17:32:44.470 -> <p>Additionally, a 400 Bad Request
17:32:44.470 -> error was encountered while trying to use an ErrorDocument to handle the request.</p>
17:32:44.470 -> </body></html>
17:32:44.470 -> 
17:32:44.470 -> 
17:32:44.470 ->  Checking for SPIFFS updates from: https://iotappstory.com/ota/esp8266-v2.0.1.php
17:32:55.454 -> <!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
17:32:55.454 -> <html><head>
17:32:55.454 -> <title>403 Forbidden</title>
17:32:55.454 -> </head><body>
17:32:55.454 -> <h1>Forbidden</h1>
17:32:55.454 -> <p>You don't have permission to access this resource.</p>
17:32:55.454 -> <p>Additionally, a 400 Bad Request
17:32:55.454 -> error was encountered while trying to use an ErrorDocument to handle the request.</p>
17:32:55.488 -> </body></html>
17:32:55.488 -> 
17:32:55.488 -> 
17:32:55.488 ->  Returning from IOTAppStory.com

cybermat, 2019-09-21:
I noticed when the "latest stable" library was installed (v2.0.2) its library.properties file contained:
name=IOTAppStory-ESP
version=2.1.0-RC1
 
After reinstalling v2.0.2 - its library.properties file now contains
name=IOTAppStory-ESP
version=2.0.2
and there are substantial differences in several files to the version v2.1.0-RC1. 
The esp8266 now gives a different error on an attempted OTA sketch update ---
18:31:15.600 -> *-------------------------------------------------------------------------*
18:31:15.600 -> Start callandroid: 
18:31:15.646 -> [HTTP] POST...
18:31:15.834 -> [HTTP] POST... failed, error: connection refused

Onno-Dirkzwager, 2020-05-03:
Hi @cybermat 
It's been some time ago.... I beleave we helped you on our Discord channel.
Are you still having issues?