iotappstory/ESP-Library #131: setClock sets the time to GMT+3

abaskin, 2019-09-24:
On the ESP8266 (or the ESP32 when SNTP_INT_CLOCK_UPD is true) setClock sets the time to GMT+3. It would seem better to set the time to GMT (as opposed to a random offset) and document that the time is set on the ESP8266 in begin and loop by default.

mcgurk, 2020-10-27:
If I understood correctly, whole setClock()-function can be removed. I commented out everything inside setClock()-function. ESP8266 API handles timezone and DST and updates time with NTP every hour. My code:
```
#include <TZ.h>
configTime(TZ_Europe_Helsinki, "pool.ntp.org", "time.nist.gov");
```
https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino

Onno-Dirkzwager, 2020-12-16:
@abaskin this is fixed with the last PR. Set to UTC. 
Why this had the strange +3 as default....I don't know....
If necessary you can change the defaults in the config.h file.

@mcgurk no need to erase code.... if you want you can turn this off in the config.h file https://github.com/iotappstory/ESP-Library/blob/master/src/config.h#L55

@abaskin @mcgurk the library needs to sync time so it can use BearSSL to perform httpS requests.

So if you turn the libraries version of time sync off, make sure you update the clock (now) yourself. Otherwise you will not be able to callhome for updates!

Can we close this issue?

mcgurk, 2020-12-17:
Now setcock() in https://github.com/iotappstory/ESP-Library/blob/master/src/IOTAppStory.cpp calls function
configTime(SNTP_INT_CLOCK_TIME_ZONE, SNTP_INT_CLOCK_TIME_OFFSET, SNTP_INT_CLOCK_SERV_1, SNTP_INT_CLOCK_SERV_2);
That is old way to set time and it doesn't support e.g. DST. 

My timezone is part of the year +2 and other part of the year +3.
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/TZ.h:
#define TZ_Europe_Helsinki	PSTR("EET-2EEST,M3.5.0/3,M10.5.0/4")

Like in example https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino time should be set with current configTime-function:
configTime(MYTZ, "pool.ntp.org"); // maybe two servers can be used

Why this is important?:
configTime handles all all time tasks. It activates ntp and saves timezone with dst-setting to "eeprom". Like wifi-functions, if it is called once and "eeprom" is not cleared, it works always even without calling configTime again in new sketch. If IOTAppStory calls configTime with fixed timezone and time offset, it overwrites user settings.

My proposal is that if setClock must be called in configTime, it uses configTime(SNTP_INT_CLOCL_TIME_ZONE, SNTP_INT_CLOCK_SERV) and SNTP_INT_CLOCL_TIME_ZONE defaults to TZ_Etc_GMT (TZ_Etc_GMT is defined as	PSTR("GMT0"), timezone 0 without DST).
After that I can modify https://github.com/iotappstory/ESP-Library/blob/master/src/config.h to define SNTP_INT_CLOCL_TIME_ZONE TZ_Europe_Helsinki

Onno-Dirkzwager, 2020-12-17:
@mcgurk thankyou for your detailed feedback!
Ill have to dive in later and will get back to you.

Onno-Dirkzwager, 2020-12-19:
@mcgurk 
First of thankyou for taking the time to give us your feedback in such a detailed fashion! (and editing multiple times)

>That is old way to set time and it doesn't support e.g. DST.

This was originally setup for the use of BearSSL. And taken from the following examples:
- [HTTPSRequest.ino#L72](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/HTTPSRequest/HTTPSRequest.ino#L72)
- [httpUpdateSecure.ino#L34](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/examples/httpUpdateSecure/httpUpdateSecure.ino#L34)

This is where the old style configTime and funny arbitrary magic numbers come from. We initially did not put enough thought into the end user.... So thankyou for making this beter!

> Like in example, My timezone, Why this is important?

Yep, you convinced me. Supporting DST is the way to go. 
>localtime()-function gives me always right local time and gmtime() gives UTC-time

Exactly what we need.

>My proposal

I fully agree and have:
- used configTime in the propossed way
- used settimeofday_cb() to update our _timeSet var
- SNTP_INT_CLOCK_TIME_ZONE now defaults to TZ_Etc_GMT
- removed SNTP_INT_CLOCK_TIME_OFFSET
- removed SNTP_INT_CLOCK_UPD_INTERVAL and the update code from our loop
- renamed ntpSync() to ntpWaitForSync()  wait for_timeSet = true   <-- we still have to wait before callhome() and going into config mode
- remove setClock() from our WiFiConnect()
- moved setClock() to above the wifi connect code as in the NTP-TZ-DST example

You can find these changes here: https://github.com/iotappstory/ESP-Library/tree/NTP-TZ-DST

>(Now I wait in Arduino setup()-function until time_is_set() is called and continue after that to IOTAppStory-calls. That way I can be sure that time is set when IOTAPPStory-functions is called.)

We / our users do not want to wait on boot. Some users don't callhome on boot and want to go ahead with whatever.... So I placed wait in the callhome and configmode parts. If settimeofday_cb() already set _timeSet = true no wait is necessary.

```
 Calling Home
 Wait for background NTP sync
 .
 GMT0: Sat Dec 19 02:24:56 2020

 Checking for App(Sketch) updates from: https://iotappstory.com/ota/updates
 No updates necessary!
```
```
 Calling Home

 Checking for App(Sketch) updates from: https://iotappstory.com/ota/updates
 No updates necessary!
```

>This is not exactly true. If configTime is ever called at least once in history and whole flash is not erased, clock is updated every hour with NTP and timezone/dst-setting are loaded from "eeprom" by ESP8266 library.

I'm probably missing / doing something wrong. Or understand you completely wrong. I feel like I tried everything.... including your suggestions in simple sketches (without our library) but I always end up with 1970 etc.

It would be of great benifit if it is posible to fall back on a previously stored time untill an ntp updates has taken place. A lot of our starting users reboot a lot in between testing. And it seems that after a while with every reboot it takes longer and longer to ntp update. This can be very enoying... especially when going into configmode:

```
 bootTimes since last update: 20
 boardMode: C
*-------------------------------------------------------------------------*
 Set NTP server settings
 - GMT0
 - pool.ntp.org
*-------------------------------------------------------------------------*
 Connecting to WiFi AP
 .....
 WiFi connected

 IP Address: 192.168.2.5
 Device MAC: F4:CF:A2:D8:45:DD
 MDNS responder started: http://iasblink-661.local
*-------------------------------------------------------------------------*
 Wait for background NTP sync
 ........................................................................................................................................................................................................ ........................................................................................................................................................................................................

 C O N F I G U R A T I O N   M O D E
 STA mode. Open 192.168.2.5
```
In extreem cases it can take nearly 2 minutes. I could fallback on compile time as this would be sufficient enought for making httpS calls in the first year or so....

Any feedback is welcome! Please check my changes and thank you for your time.

https://github.com/iotappstory/ESP-Library/tree/NTP-TZ-DST

mcgurk, 2020-12-19:
Thank you for your nice replies.

Here is post from arduino-forum which was my eye opener how deeply NTP is in ESP8266 API:
https://forum.arduino.cc/index.php?topic=655222.0
But that and my example was little sidetrack and impractical, because there is no reason to try do sketch without calling configTime at all. Maybe it doesn't even work like that with current API/libraries.

I noticed too that sometimes it gets longer time to get NTP-time. I also use "change something in code - compile and upload - test - reboot"-method multiple times in a row in short time span, but NTP sync never took so long that it would be too annoying for me.

I don't know any straightforward sollution for remembering time after reboot, but if I understood correctly, ESP8266 RTC-memory survives deepsleep and reboot. Every time time_is_set-function is called, new time could be saved to RTC-memory. And if that is set to ESP8266 internal time in setup() (or in library-setup-functions), time can be only one hour off at most after reboot/deepsleep.
https://www.youtube.com/watch?v=r-hEOL007nw

Onno-Dirkzwager, 2020-12-20:
@mcgurk I made some more changes as this has to be compatible with the ESP32. It's not turned on by default for the ESP32 as we do not need it for certificate validation. But some of our users use it for their projects...

Their are some differences between the platforms:

- ESP32 does not support the callback settimeofday_cb
- ESP32 does not come with the TZ.h definitions file
- ESP32 does support setting configTime with a timezone
- settimeofday is set slightly different between the two.

We want to keep the library as free as possible from platform definitions whithin the main code. So I moved the ntp code to separate files and let the include.h decide which to include.

- ["espressif/esp32/NtpTimeSync.h"](https://github.com/iotappstory/ESP-Library/tree/NTP-TZ-DST/src/espressif/esp32)
- ["espressif/esp8266/NtpTimeSync.h"](https://github.com/iotappstory/ESP-Library/tree/NTP-TZ-DST/src/espressif/esp8266)

>I don't know any straightforward sollution for remembering time after reboot, but if I understood correctly, ESP8266 RTC-memory survives deepsleep and reboot.

Yes I thought about this to as we use it for storing modes (N & C) and bootimes. I have not implemented this yet. But have included Settimeofday functions in the files mentiond above. So users can already use them if they want to. And we may decide to in the future.

Please test the [NTP-TZ-DST](https://github.com/iotappstory/ESP-Library/tree/NTP-TZ-DST) branche and let me know if we can close this issue and add the code to the main branche.

Onno-Dirkzwager, 2020-12-24:
@mcgurk 
added the "ESP8266 RTC-memory" solutions as we talked about. This drastically brings down the wait time in slow NTP sync situations.

https://github.com/iotappstory/ESP-Library/blob/NTP-TZ-DST/src/espressif/esp8266/NtpTimeSync.cpp#L31

closed 2020-12-24 by Onno-Dirkzwager