iotappstory/ESP-Library #132: Own project implementation possible?

exocode, 2019-09-26:
Hi there, 

this library is so awesome 👍 . I love the way how I can select projects from an interface within the Thing WebServer and provision the Things.

Is it somehow possible to implement/change the behavior so, that this library is interacting with my own API project (different domain)?

If yes, how/where can I start? 
- Which API endpoints do I need? 
- And what should the response of these endpoints?
- or in other words, is there a repo/project where I can see how it's interacting?

If no, you may suggest a similar library?

Many, many thanks in advance

Gruezi :-)

CwlBroeders, 2019-09-28:
That would depend on what behavior u are looking for

exocode, 2019-09-29:
Hi @CwlBroeders, thank you for your response. 

I am quite new on all that stuff (OTA).

What I try to do / what I am searching for:

- Something similar to what "iotappstory.com" does in conjunction with this library.
- closed ecosystem (I wanna offer "programs" and devices together)
- users can not upload/share their own projects
- I wanna use Arduino driven projects with specific functions

I am more a frontend and API developer (I experiment with "Things" and C++ since this year, but I found out that it is too much time-consuming and I stuck in so many tasks. Maybe a more experienced C++ developer would attend my idea/project / could help me out of this "hell".

Behavior what I am looking for:
- As a User, I create an account on myPage
- A user orders pre-configured things on myPage
- A user buys program-receipts on myPage
- A user connects the device and can attach/load a receipt on his device
- A user can watch reports and change the behavior of his device via OTA or Thing web interface (like yours does)
 
It would be great if
- there would be a "manual" how to provision things 
- how I can use that library on my backend/API
- which API endpoints exist and in which order all that happens
- an open-source "iotAppStory" repo :-)

I found something similar here: http://thinx.cloud

I am happy for any suggestion :-)

Thank you

Onno-Dirkzwager, 2020-05-03:
Hi @exocode,
I'm not entirely sure if we can help but if you are still interested please contact me at iotappstory@gmail.com We are helping some other proffessional makers so who knows!

Im closing this issue as it is not an code issue. And we will chat elsewhere! But feel free to reopen it if deemed necessary.

closed 2020-05-03 by Onno-Dirkzwager