iotappstory/ESP-Library #133: Config Page Blacked out.

MigalBothma, 2019-10-02:
Model : Lolin v3 ESP8266 ( NodeMCU12E )

IOTAppStory Sketch : IASLoader from Example ( Library version : 2.1.0-RC2 )
ESP8266 Arduino Core ( 2.5.2 )  

Syptoms : Config JS page on my local network (10.0.0.11) is black.

Errors in console ( Chrome with DevTools installed ):
----------------------------------------------------------------------------------------------------------

10.0.0.11/:1 A cookie associated with a cross-site resource at http://iotappstory.com/ was set without the `SameSite` attribute. A future release of Chrome will only deliver cookies with cross-site requests if they are set with `SameSite=None` and `Secure`. You can review cookies in developer tools under Application>Storage>Cookies and see more details at https://www.chromestatus.com/feature/5088147346030592

VM318:1 Uncaught SyntaxError: Unexpected token < in JSON at position 0
    at JSON.parse (<anonymous>)
    at cfg.js:211
    at cfg.js:775
    at XMLHttpRequest.xhttp.onreadystatechange (cfg.js:712)

Uncaught SyntaxError: Unexpected token < in JSON at position 0
    at JSON.parse (<anonymous>)
    at cfg.js:211
    at cfg.js:775
    at XMLHttpRequest.xhttp.onreadystatechange (cfg.js:712)

----------------------------------------------------------------------------------------------------------

Upon further inspection of the scripts at line 211 it fails the code :
		tmplLoader("pagelayout", "", function(response){		
			response = JSON.parse(response);

The response is returned and the pagelayout received like :
----------------------------------------------------------------------------------------------------------
<br />
<b>Warning</b>:  include_once(): open_basedir restriction in effect. File(/home/iotappstory/config/config.inc.php) is not within the allowed path(s): (/home/iotapp/:/tmp/:/var/tmp/:/opt/alt/php`PHP1_RELEASE`/usr/share/pear/:/dev/urandom:/usr/local/php72/lib/:/usr/local/php72/lib/:/usr/local/lib/php/) in <b>/home/iotapp/domains/iotappstory.com/public_html/ota/config/cfg.php</b> on line <b>15</b><br />
<br />
<b>Warning</b>:  include_once(/home/iotappstory/config/config.inc.php): failed to open stream: Operation not permitted in <b>/home/iotapp/domains/iotappstory.com/public_html/ota/config/cfg.php</b> on line <b>15</b><br />
<br />
<b>Warning</b>:  include_once(): Failed opening '/home/iotappstory/config/config.inc.php' for inclusion (include_path='.:/usr/local/php72/lib/php') in <b>/home/iotapp/domains/iotappstory.com/public_html/ota/config/cfg.php</b> on line <b>15</b><br />
<br />
<b>Warning</b>:  include_once(): open_basedir restriction in effect. File(/home/iotappstory/system/xtra.func.php) is not within the allowed path(s): (/home/iotapp/:/tmp/:/var/tmp/:/opt/alt/php`PHP1_RELEASE`/usr/share/pear/:/dev/urandom:/usr/local/php72/lib/:/usr/local/php72/lib/:/usr/local/lib/php/) in <b>/home/iotapp/domains/iotappstory.com/public_html/ota/config/cfg.php</b> on line <b>18</b><br />
<br />
<b>Warning</b>:  include_once(/home/iotappstory/system/xtra.func.php): failed to open stream: Operation not permitted in <b>/home/iotapp/domains/iotappstory.com/public_html/ota/config/cfg.php</b> on line <b>18</b><br />
<br />
<b>Warning</b>:  include_once(): Failed opening '/home/iotappstory/system/xtra.func.php' for inclusion (include_path='.:/usr/local/php72/lib/php') in <b>/home/iotapp/domains/iotappstory.com/public_html/ota/config/cfg.php</b> on line <b>18</b><br />
<br />
<b>Warning</b>:  include_once(): open_basedir restriction in effect. File(/home/iotappstory/lib/core/pdoQuery.php) is not within the allowed path(s): (/home/iotapp/:/tmp/:/var/tmp/:/opt/alt/php`PHP1_RELEASE`/usr/share/pear/:/dev/urandom:/usr/local/php72/lib/:/usr/local/php72/lib/:/usr/local/lib/php/) in <b>/home/iotapp/domains/iotappstory.com/public_html/ota/config/cfg.php</b> on line <b>19</b><br />
<br />
<b>Warning</b>:  include_once(/home/iotappstory/lib/core/pdoQuery.php): failed to open stream: Operation not permitted in <b>/home/iotapp/domains/iotappstory.com/public_html/ota/config/cfg.php</b> on line <b>19</b><br />
<br />
<b>Warning</b>:  include_once(): Failed opening '/home/iotappstory/lib/core/pdoQuery.php' for inclusion (include_path='.:/usr/local/php72/lib/php') in <b>/home/iotapp/domains/iotappstory.com/public_html/ota/config/cfg.php</b> on line <b>19</b><br />
<br />
<b>Notice</b>:  Undefined variable: DBhost in <b>/home/iotapp/domains/iotappstory.com/public_html/ota/config/cfg.php</b> on line <b>23</b><br />
<br />
<b>Notice</b>:  Undefined variable: DBName in <b>/home/iotapp/domains/iotappstory.com/public_html/ota/config/cfg.php</b> on line <b>23</b><br />
<br />
<b>Notice</b>:  Undefined variable: DBuser in <b>/home/iotapp/domains/iotappstory.com/public_html/ota/config/cfg.php</b> on line <b>23</b><br />
<br />
<b>Notice</b>:  Undefined variable: DBpass in <b>/home/iotapp/domains/iotappstory.com/public_html/ota/config/cfg.php</b> on line <b>23</b><br />
<br />
<b>Fatal error</b>:  Uncaught PDOException: SQLSTATE[HY000] [2002] No such file or directory in /home/iotapp/domains/iotappstory.com/public_html/ota/config/cfg.php:23
Stack trace:
#0 /home/iotapp/domains/iotappstory.com/public_html/ota/config/cfg.php(23): PDO-&gt;__construct('mysql:host=;dbn...', NULL, NULL)
#1 {main}
  thrown in <b>/home/iotapp/domains/iotappstory.com/public_html/ota/config/cfg.php</b> on line <b>23</b><br />

----------------------------------------------------------------------------------------------------------

What I'm trying now:
I had a different ArduinoJSON library installed. Will try to uninstall potential library mismatch.

CwlBroeders, 2019-10-02:
Thanks for telling us, @onno is working on it but it's not a quick fix, please refrain from using the RC versions for now

MigalBothma, 2019-10-08:
Thanks for the answer provided.

I reverted back to the stable version (2.0.2) with Arduino ESP8266 Core(2.4.2)

I am humbly impressed with the progress you guys have shown in this project! 

You also did confirm that the server was offline on Discord. 10/10 Support!

closed 2019-10-08 by MigalBothma