iotappstory/ESP-Library #136: Cant Upload IASloader example code to NodeMCU1.0

sridharblue, 2020-01-03:
Arduino: 1.8.10 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled (new can abort), All SSL ciphers (most compatible), 4MB (FS:1MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Multiple libraries were found for "ESP8266WiFi.h"
 Used: C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WiFi
Multiple libraries were found for "EEPROM.h"
 Used: C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\EEPROM
Multiple libraries were found for "ESPAsyncTCP.h"
 Used: C:\Program
Multiple libraries were found for "ESP8266HTTPClient.h"
 Used: C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266HTTPClient
Multiple libraries were found for "DNSServer.h"
 Used: C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\DNSServer
Multiple libraries were found for "IOTAppStory.h"
In file included from C:\Users\Administrator\Documents\Arduino\libraries\IOTAppStory-ESP\src\espressif\updateNextion.cpp:2:0:

 Used: C:\Users\Administrator\Documents\Arduino\libraries\IOTAppStory-ESP
C:\Users\Administrator\Documents\Arduino\libraries\IOTAppStory-ESP\src\espressif\updateNextion.h:6:27: fatal error: ESPNexUpload.h: No such file or directory

Multiple libraries were found for "ESPAsyncWebServer.h"
  #include <ESPNexUpload.h>

 Used: C:\Program
                           ^

Multiple libraries were found for "Hash.h"
compilation terminated.

 Used: C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\Hash
Multiple libraries were found for "ESP8266mDNS.h"
 Used: C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266mDNS
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

OnJuanTrack, 2020-01-03:
I'm having the same problem, I've traied install the 2.4.2 ESP8266 library and it does not works 

/home/XXX/Arduino/libraries/IOTAppStory-ESP/src/espressif/esp8266/callServer_WiFiClientSecure.cpp:167:13: error: 'class axTLS::WiFiClientSecure' has no member named 'setFingerprint'
     _client.setFingerprint(_config->sha1);

CwlBroeders, 2020-01-03:
@sridharblue 
Looking into it, if u have no need for nextion, as we discussed on discord, u could also comment out lines 38,39,40 in libraries/IOTAppStory.src/includes.h

https://github.com/iotappstory/ESP-Library/blob/master/src/includes.h#L38

@OnJuanTrack 
If u are using the 2.1.0 RC version of the library u will need to use the 2.5.2 core version or higher (IAS tested for 2.5.2)

Alternatively u could also use IAS version 2.0.2
https://github.com/iotappstory/ESP-Library/tree/2.0.2

OnJuanTrack, 2020-01-03:
@sridharblue  now this code is working for me, to solve this you have to do this:

IOTAppstory library: https://github.com/iotappstory/ESP-Library/releases/latest

ESP8266 specific
ESP8266 core for Arduino >= 2.4.2
Async TCP Library for ESP8266
Arduino IDE >= 1.8.5
Async Web Server

ESP32 specific
Arduino core for the ESP32 >= 1.0.0
Async TCP Library for ESP32
Arduino IDE >= 1.8.5
Async Web Server

this works for me (ESP8266)

closed 2020-01-06 by sridharblue