iotappstory/ESP-Library #137: WLAN connection failed

OnJuanTrack, 2020-01-03:
Hi friends,

I'm traying to use the IASLoader, when I enter into the config page (192.168.4.1) and I entry my SSID and password always I receive " Failed, try again!!!", I've probed with different WLAN (My current wifi and an AP from my phone) but always I have this message, I'm sure that my password is correct then I don't understand what is happening.

I'm using an ESP8266, with default configuration 115200, 80Mhz, 4M, disable, ninguno, v2 lower memory, flash, disable, only sketch

My IDE version:  1.8.10 
IOTAppStory-ESP version: 2.0.2
Async TCP Library 
Async Web Server
ESP8266 library board: 2.4.2

 Thanks friends.

CwlBroeders, 2020-01-03:
@OnJuanTrack 
There is a known problem with special characters, this could be the cause, but i dont know your wifi credentials ofcourse :) 

https://github.com/iotappstory/ESP-Library/issues/106

OnJuanTrack, 2020-01-03:
Hi friend, my current WIFI have some special characters, but when I create my AP with my phone I use the password: "holahola" and even in this way this does not connect.

CwlBroeders, 2020-01-03:
Interresting, this part of the library usually just works :)

a few questions to see if we can find what the problem may be.

Have u flashed the board with the IAS 2.1.0. RC version at an earlier moment?

Have u made any changes to the config.h file regarding wifi?
https://github.com/iotappstory/ESP-Library/blob/2.0.2/src/config.h#L18

closed 2020-01-07 by OnJuanTrack