iotappstory/ESP-Library #138: Max device limit

OnJuanTrack, 2020-01-07:
Hi IOTAppStory, in this moment I'm working in a project where we're using your platform (I wanna say you that this platform is incredible), in our project we have to handle maybe 500 or 1000 devices, so we wanna know how can we do it.

Also I wanna know if is possible have more than 2880 callhome because if I have to handle 1000 devices, this number is not enough

CwlBroeders, 2020-01-07:
These limits only apply to free accounts.

OnJuanTrack, 2020-01-08:
Where can I know some prices to buy a pay account?

Impatience71, 2020-04-15:
Dear support team,  
I would also be happy to know, how it runs with more devices in your platform. I mean are there different price models for different amount of devices, different service levels ? 
I am excited about the idea and the implementation so far. Very cool. 
Br Peter

Onno-Dirkzwager, 2020-05-03:
@Impatience71 please contact us at iotappstory@gmail.com
Im closing this issue as @OnJuanTrack and I already spoke on other channels.

closed 2020-05-03 by Onno-Dirkzwager