iotappstory/ESP-Library #140: mDNS resolution fails for IASBlink

jesseparker, 2020-02-08:
I am using the latest code from ESP-Library/master with Arduino 1.8.5 and .10 and ESP-8266 2.5.* on the LOLIN(Wemos) D1 R2 & mini board.

The sketch outputs to serial that mDNS is working and gives local name, but my Linux system running Avahi 0.6.31 or my Windows 10 system cannot resolve the name.

For both my Linux system and Windows system I can reliably resolve a local host name through mDNS when I upload the ESP8266mDNS example mDNS_Web_Server while using the same development setup.

I have traced this problem through to the ESP8266mDNS library and tried the current LEAmDNS implementation and "Legacy".  In both cases the mDNS library receives the multicast query, however decides a reply is not required.  If I trick the code into replying, the hostname still doesn't resolve.  I am using linux mdns-scan and I don't see the device, but I also don't see any device when the mDNS_Web_Server example is working successfully.

At this point I've come to the limit of my ESP8266 understanding...  I don't know how to run a debugger in-process for 8266.

This issue greatly impacts the usefulness of a device based on IOTAppStory, because you can't provide someone with a device that says on it "plug me in and vist http://foo.local to configure"  The option is to go looking for the device on your router's DHCP server and then correctly enter the IP address - this is beyond the comfort zone of normal people.

I have a device I am releasing through CCA license on thingiverse that is an improvement of https://www.thingiverse.com/thing:4094036 but uses a length of pipe as a chime - It's like a grandfather clock that "chimes the hour" and it's a nice device to have in my home.  If you can fix this issue I promise to release an open source 3d printed grandfather clock like device that requires only a D1 mini, hobby servo, a piece of pipe, and 3 printed parts and base it on IOTAppStory.  I think your approach is great and a missing piece of the puzzle for simple IOT development, but for "consumer" users a few more improvements would improve the suitability of your product a lot.

jesseparker, 2020-02-28:
I'd really like to release my device as an IotAppStory device, but this mDNS issue kind of makes it harder than just compiling code yourself...  https://youtu.be/eZ2nAdM6e6I

Onno-Dirkzwager, 2020-05-02:
Hi @jesseparker ,
Sorry for the late reply! @CwlBroeders thanks for notifying me.

Quite some time has passed since your initial issue... Are you still working on this project?

I would like to help you and started testing as we speak.

Onno-Dirkzwager, 2020-05-03:
@jesseparker 
During tests I found out that the ESP32 had no problems mDNS wise compaired to the ESP8266. Turns out we were missing MDNS.update(); for ESP8266 devices (and ESP32 devices don't need it)

Although this will help. (And I'm happy you pointed us to this bug)
I'm not entirely sure this will solve everything as mDNS is not always supported and needs third party tools / plugins: https://stackoverflow.com/questions/23624525/standard-mdns-service-on-windows

I would be more than happy to brainstorm with you about solutions for easy access to the end users.

Onno-Dirkzwager, 2020-05-16:
@jesseparker just released [Release Candidate nr 3](https://github.com/iotappstory/ESP-Library/releases)

Onno-Dirkzwager, 2020-05-18:
@jesseparker any chance you can test if this solves your issue?

Onno-Dirkzwager, 2020-06-08:
@jesseparker Im going to close this issue due to lack of feedback and confirmation of other people. Feel free to reopen if deemed necessary!

closed 2020-06-08 by Onno-Dirkzwager