iotappstory/ESP-Library #141: Error Compiling for board TTGO T1 or any Esp32 board

nasserbek, 2020-05-01:
Hello,
I'm trying to flash my esp32 board with the IASLoader.ino and I have always this error during compliation, I followed the instructions on the IOTAPPStory.com for library 2.2 esp32 last version and still facing the issue...

Thanks for any help
Regards

CwlBroeders, 2020-05-01:
Was this the only error message the compiler gave?
If not, please post the rest of the errors aswell.

And what are your board settings?

nasserbek, 2020-05-01:
Thanks for your replay..

I have only this error when I compile:
exit status 1
Error compiling for board TTGO T1.

Also under File-Examples-Incompatible I find the IASLoader and other examples of IOTAPPStory.. Why incompatible!!

My actual settings are:

Board:TTGO T1
CPu freq: 240 Mhz
Flash mode:QIO
Flash Size 4 MB(32MB)
Partition scheme:Default 4 Mb with spiffs (1.2 Mb APP/1.5 MB Spiffs)
Core debug:none

Thanks for your support

CwlBroeders, 2020-05-01:
Just to make sure, this is for IAS version 2.0.2, not the 2.1.0-RC2?

nasserbek, 2020-05-01:
Tried both versions 2.0.2 and 2.1.0-RC2 with esp32 espressive version 1.0.4 and still have the error.

I'm using Arduino 1.8.10 with Windows 10 maybe here is the problem, what OS you are using?

If it works for you this is good news which means that the problem is in my configuration..
I'll try to have same configuration like you and let you know..

Regards

nasserbek, 2020-05-01:
Tried with Arduino 1.8.9 and still not working, could be windows 10 the cause?
now I have:
Arduino 1.8.9
Esp32 version 1.0.4
Library 2.0.2

I have the errore with any esp32 board..

CwlBroeders, 2020-05-01:
looks like everything is the same for us.

Could u please copy every single line from the compiler feedback

nasserbek, 2020-05-01:
Here is the log:

Arduino: 1.8.9 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

In file included from E:\Onedrive\Arduino\libraries\libraries\ESP-Library-2.0.2\examples\IASLoader\IASLoader.ino:33:0:

E:\Onedrive\Arduino\libraries\libraries\ESP-Library-2.0.2\src/IOTAppStory.h:14:25: fatal error: WiFiMulti.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Thanks for your support..
Regards

CwlBroeders, 2020-05-01:
Looks like we found a clue

E:\Onedrive\Arduino\libraries\libraries\ESP-Library-2.0.2\src/IOTAppStory.h:14:25: fatal error: WiFiMulti.h: No such file or directory

The compiler is unable to find the WiFiMulti.h file.

This is strange because this code should come with the Esp32 version 1.0.4, by default.
https://github.com/espressif/arduino-esp32/blob/1.0.4/libraries/WiFi/src/WiFiMulti.h

Could u check if the file exists on your system?

nasserbek, 2020-05-01:
Yes, I noticed that but I have the WiFiMulti.h file under this folder:

C:\Program Files (x86)\Arduino\hardware\Expressif\Esp32\libraries\WiFi\src

Regards

CwlBroeders, 2020-05-01:
For some reason the compiler is unable to recognize it at such, i have seen similar problems in the past.

Could u please remove the ESP32 with the IDE board manager, then close the IDE and check if the files have been removed and open up de IDE again and re install the ESP32 board thru the manager

nasserbek, 2020-05-01:
I reoved the esp32 and removed all the folder:
 C:\Program Files (x86)\Arduino\hardware\Expressif\Esp32\libraries

Then installed the esp32 core 1.0.4 again but now I have this error:

AsyncTCP.h: No such file or directory

this file is not present on my computer, I downloead from github bnd but it under the library 2.0.2 , compiled bu tget a new file missing :

fatal error: ESPAsyncWebServer.h: No such file or directory

Regards

nasserbek, 2020-05-01:
Also the file ESPAsyncWebServer.h was not present on my PC, downloaded and installed it under library and now I have another error:
fatal error: StringArray.h: No such file or directory

This file also is not present on my PC.

Regards

nasserbek, 2020-05-01:
I downloaded 3 librarries from the web, installed them and finally I can complie without errors:
-AsyncTCP-master
-ESPAsyncWebServer-master
-AsyncTCP-master

Without theses librarieries it would never work , I believe you have them already installed on your PC.

Thanks a lot for your kind and prompt support...I relly appreciate..

closed 2020-05-01 by nasserbek