iotappstory/ESP-Library #155: Details of use?

jmoudy, 2020-05-18:
Loving IOTAPPSTORY ... I have a few questions.

- Does this application powerdown the Wifi when it is not being used to check for updates? 
- I know the wifi library calls are available, but can I power down the wifi without causing a problem for IOTAPPSTORY? 
- Also, if so, can I bring up the wifi system, then make a call to check for update, then I'll power down the wifi?

Onno-Dirkzwager, 2020-05-19:
@jmoudy 
- The library does not power down Wifi
- Yes you can! It is wise to take full controle... don't use the update interval but only [callhome](https://iotappstory.com/wiki/arduino-esp/library-api-20/api-methods/callhome()) when your conditions are met.... awake, connected etc.

Onno-Dirkzwager, 2020-05-27:
@jmoudy does this answer your question? If so please close the issue.

Onno-Dirkzwager, 2020-06-03:
@jmoudy closing issue due to lack of response.
Feel free to reopen if deemed necessary

closed 2020-06-03 by Onno-Dirkzwager

jmoudy, 2020-06-03:
Thanks, I'll work on getting an example built. Waiting for new hardware. 

Jim