iotappstory/ESP-Library #156: Incompatibility with UniFi AP?

btkramer, 2020-05-21:
There seems to be a compatibility issue with my AP, which is a Ubiquiti Unifi AP AC LR, the model is UAP‑AC‑LR, it's 802.11ac

Signal is strong, no congestion.
I have no problem getting online with a standard wifi OTA and wifi manager sketch and this AP.

The ESP appears to freeze after credentials are saved and a connection attempt is made to the AP. After changing the wifi multi setting to false in config.h I can enter the wifi credentials and it will connect, but only after restarting the ESP. It still freezes at the same point of entering credentials.

I also have no problem with this library and my phone's hotspot. Everything works as intended in this case.

Onno-Dirkzwager, 2020-05-22:
@btkramer 
Please check if #106 has anything to do with your ssid / password.

btkramer, 2020-05-22:
@Onno-Dirkzwager 
SSID is plain letters with 2 spaces (3 words with spaces between them), no other special characters. password is numbers only.

SiskoBen, 2020-05-22:
Ever since i started with IOT AppStory use i've been having a Ubiquity Unifi AP AC LR and never had issues described here. I'm currently on v4.0.80.10875 on the AP AC LR. I am using AppStory v.2.1.0RC3 under Platform IO using platform v2.5.1 which uses core 2.7.1 the devices i use are mostly Wemos D1 mini (both original and knock offs). Hope this helps.

Onno-Dirkzwager, 2020-05-22:
@btkramer @btkramer thankyou for the feedback.

@SiskoBen points to something I have had a few times myself. "copy the SSID from the list by adding a space behind it"

Because my phone reconizes the "word" Im typing from previous sessions. It often finishes the word and adds a space behind it.

Might be worth checking.

btkramer, 2020-05-22:
@Onno-Dirkzwager @SiskoBen

I am on v4.3.13 on my AP. I can confirm there are no spaces before the ssid on credential entry. after changing wifi multi to false in config.h, it now does connect to the AP before freezing, as apparent in the AP's logs, but the ESP does freeze at this point. then after restarting the ESP, it connects to the AP as normal and you can then navigate to the IAS config page

Onno-Dirkzwager, 2020-05-22:
@btkramer 
Well at least you can then continue adding your device, certificate and doing OTA updates. 
While we figure this out together. Do you have an extra esp to keep asside for experiments regarding this issue?

btkramer, 2020-05-22:
@Onno-Dirkzwager 
Yeah, I have a few spares :) I am actually mostly interested in this project for the ability to remote update Nextion devices, which it appears you headed up, and serial monitor over wifi. Those things were on my list to integrate into my projects, but I hadn't gotten around to yet.

Onno-Dirkzwager, 2020-05-24:
@btkramer how often does your callhome end in: Error: connection failed?

CwlBroeders, 2020-05-27:
Is this issue still unresolved?

Onno-Dirkzwager, 2020-05-27:
Yes, @btkramer has found an workaround to continu. But the main issue is still unresolved.

> after changing wifi multi to false in config.h, it now does connect to the AP before freezing, as apparent in the AP's logs, but the ESP does freeze at this point. then after restarting the ESP, it connects to the AP as normal and you can then navigate to the IAS config page